LibreCAD/LibreCAD #1247: ANSI B size problem

slackuser0xae34, 2020-06-11:
ANSI B Issues:
	1) Export ANSI B size to PDF fails, but the other ANSI sizes export correctly.
	2) Page size in "Options" -> "Current Drawing Preferences" -> "Paper" tab revert to "Ledger".

Built from LibreCAD-master (of 2020/06/06, a857c382e316847aeb1a398cbc5713af1489b444)
	- Export to PDF for ANSI B (only) size fails. The paper is oriented in portrait orientation, when it should be in landscape orientation. However, the drawing image is oriented correctly (in landscape).

### Steps to reproduce or sample file
1) Create a drawing, using "ANSI B" in "Options" -> "Current Drawing Preferences" -> "Paper" tab -> "Format" : "ANSI B" -> "OK"
2) Draw a rectangle:
	a) "Command" box, Enter: "rect"
	b) "Specify first corner", Enter: 1,1
	c) "Specify second corner", Enter: 16,10
	d) Press "Escape" to end drawing rectangles.
3) Click on "Auto Zoom" button.
4) Click on "Print Preview" button.
5) In the "Print preview" window
	a) Set the scale to "1:1"  (Uncheck the "fixed" box, if necessary).
	b) Click the "Fit to page" button (Uncheck the "fixed" box, if necessary).
	c) Click the "Center to page" button (Uncheck the "fixed" box, if necessary).
	d) Set the "fixed" box checked.
6) Verify that a rectangle appears on the print preview window. 
	a) The rectangle should be centered in the window.
	a) There should look like a "margin" around the rectangle. 
7) Export to PDF in "File" -> "Export" -> "Export as PDF"
	a) Select a directory and enter a file name for the PDF output.
	b) Click the "Save" Button.
	c) Verify that the saved PDF file exists in the selected directory.
8) In a PDF viewer application (outside LibreCAD), view the saved PDF file.

### Expected behavior
1) Steps 1) to 8) as above
2) In a PDF viewer application,a landscape image should be centered, and completely contained, in a landscape oriented page (wider than it is tall).
3) In LibreCAD, "Options" -> "Current Drawing Preferences" -> "Paper" tab -> "Format" -> "ANSI B"

### Observed behavior
1)  Steps 1)to 8) as above
2) In a PDF viewer application, the page is oriented wrong, with the image oriented correctly.
	a) The page is oriented in "portrait" orientation, (taller than it is wide).
	b) The image is oriented in "landscape" orientation, but partially cut off where it leave the page on the right side.
3) In LibreCAD, "Options" -> "Current Drawing Preferences" -> "Paper" tab -> "Format" -> "Ledger"

### Operating System and LibreCAD version info
OS: Linux
	a) Slackware 14.2
	b) QT 5.9.8 (nowebengine)
	c) MuParser 2.2.5
LibreCAD
	a) LibreCAD-master (from Github)
	b) Identifier: a857c382e316847aeb1a398cbc5713af1489b444
	c) Date: 6/6/20, 6.23 PM

### Additional Notes
An additional apparent error occurs:

1) Set up the drawing for ANSI B and draw something.
2) Go to the print preview window:
	a) The fixed box is unchecked
	b) Press the "Fit to page" button.
	c) In the "Command line" window, the following error message appears
...
Warning:: Paper size less than 10mm. Paper is too small for fitting to page
Please set paper size by Menu: Edit->Current Drawing Preferences->Paper
...

3) Export to PDF as above. The PDF is incorrect as indicated above. I don't know if this is related to ANSI B only or goes further (I haven't had time to check yet).

#####

During preliminary trouble shooting (which is incomplete):

1) In rs_units.cpp, changing the dimensions of "ANSI B" from being identical to "Ledger" (279.4, 431.8) to values smaller by 0.1 millimeter (279.3, 431.8) then the "Page Size" reporting is correct (ANSI B Issue #2 above). 
	Note, there are a couple of side effects of this. The "Options" -> "Current Drawing Preferences" -> "Paper" tab -> "Format" -> "Height" is off by 4/1000 (17.000 x 10.996 versus 17.000 x 11.000). Also, the scaling on the print preview window is a bit off.
	However, in the exported PDF file, the paper is still oriented incorrectly, with the image oriented corrctly (ANSI B Issue #1 above).

2) The fix in bug report 1229 appears mostly correct (in that it works similarly to the suggested patch in bug report 1140). Patch 1140 worked in earlier versions of LibreCAD-master.

The fix for bug report 1229 doesn't appear to address the following QT ideosyncracy (or perhaps it addresses it in another way which I haven't figured out, but doesn't appear to work):

From "qt-everywhere-opensource-src-5.9.8/qtbase/src/gui/painting/qpagesize.h"

<QT extract>

static const StandardPageSize qt_pageSizes[] = {
     ...

    {QPageSize::Letter            ,   DMPAPER_LETTER               ,   QPageSize::Inch      ,    612,  792,      215.9,  279.4,       8.5 ,  11   ,    "Letter"},
    {QPageSize::Legal             ,   DMPAPER_LEGAL                ,   QPageSize::Inch      ,    612, 1008,      215.9,  355.6,       8.5 ,  14   ,    "Legal"},
    {QPageSize::Ledger            ,   DMPAPER_LEDGER               ,   QPageSize::Inch      ,   1224,  792,      431.8,  279.4,      17   ,  11   ,    "Ledger"},
    {QPageSize::Tabloid           ,   DMPAPER_TABLOID              ,   QPageSize::Inch      ,    792, 1224,      279.4,  431.8,      11   ,  17   ,    "Tabloid"},
    {QPageSize::AnsiC             ,   DMPAPER_NONE                 ,   QPageSize::Inch      ,   1224, 1584,      431.8,  558.8,      17   ,  22   ,    "AnsiC"},
    {QPageSize::AnsiD             ,   DMPAPER_NONE                 ,   QPageSize::Inch      ,   1584, 2448,      558.8,  863.6,      22   ,  34   ,    "AnsiD"},
    {QPageSize::AnsiE             ,   DMPAPER_NONE                 ,   QPageSize::Inch      ,   2448, 3168,      863.6, 1118  ,      34   ,  44   ,    "AnsiE"},

    ...
    }

    ...

    The defined size is always in width x height order with no implied page
    orientation. Note that it is possible for page sizes to be defined where the
    width is greater than the height, such as QPageSize::Ledger, so you cannot
    rely on comparing the width and height values to determine page orientation.
    ...
    \value AnsiA = Letter
    \value AnsiB = Ledger
    \value AnsiC
    \value AnsiD
    \value AnsiE
    ...

</QT extract>

Note in the above that "Tabloid" is oriented similarly to "Letter/AnsiA", "AnsiC", "AnsiD", "AnsiE". Whereas "AnsiB" is mapped to "Ledger", which is oriented uniquely. 

###

Further, it appears that some other change in LibreCAD has affected the portrait/landscape code and may be the cause of the failure to orient the paper correctly for ANSI B. I observe this from successfully using the suggested 1140 patch (mostly identical to the 1229 patch) in a slightly older version of LibriCAD (Github: LibreCAD-master (574d9f0..., dated 5/21/20). Ergo, another change implemented between 2020/05/21 and 2020/06/06 has changed the handling of the page size and orientation, affecting the "ANSI B" handling.

I'm still digging. For those who don't mind a small laugh, it is my lot in life that while I use all the ANSI sizes occassionally, I use ANSI B the most frequently! ANSI B size drawings can usually reduce acceptably to letter size paper; the larger ANSI size drawings printed on letter size paper become so small as to become hard to read. Where as Letter/ANSI A size drawings usually require more sheets, and are hard to keep track of in order to understand the content/intent of the drawing.

Thanks all.

GSStnb, 2020-06-12:
I've been able to address 1) - I used "ledger" with the "US Office" sizes when I should have used "tabloid".  One of the standards I referred to stated "tabloid" is the same size as "ledger", but the orientation is portrait whereas ledger is landscape - and I'm guessing it is also that way in the library.  After a quick test, changing it to "tabloid"  resolves the problem.
It sort of resolves 2) - it now reverts to "Tabloid" :-)
Both 1 & 2 can be resolved by removing "Tabloid" (Ledger) from  the Drawing Preferences options, but the same thing happens with ANSI-A (it reverts to "Letter").  For those not familiar with ANSI sizes, that might be a problem.
I'll push it once I test to make sure it works properly.
(I use ANSI-B a lot too.)

slackuser0xae34, 2020-06-13:
GSStnb,

Many thanks for the quick response. I've done further testing and agree with you on the use of "Tabloid" (in lc_printing.cpp). This appears to be a quirk of QT5.

I've also seen the ANSI A reverting to letter, in my latest testing. Below are my results and my suggested patch.

I suggest that "Letter" and "ANSI A" be combined into "Letter/ANSI A", also that "Ledger" and "ANSI B" be combined into "Ledger/ANSI B". I further suggest that the separate "ANSI A" and "ANSI B" be eliminated. (I a bit verbose below, forgive me).  I've coded up my suggestions and they appear to work, to my current level of testing. The diff file is below. Thanks. S.

#####
ANSI B Issues (2020/06/13 update):
1) Export ANSI B size to PDF fails, but the other ANSI sizes export correctly.
2) Page size in "Options" -> "Current Drawing Preferences" -> "Paper" tab revert to "Ledger".

Did you ever have one of those periods where you believed that the universe was laughing at you behind your back. Well I'm having one.

I've done some more troubleshooting on LibreCAD-master (0e463c42db0b0b26dfcc230a20c3596b53f26f5e, of 2020/06/11). I have found some possible solutions to the two problems above. I haven't addressed the odd message in the command line box.

Item 1), Export ANSI B:
The ANSI B PDF export appears to be the Ledger/Tabloid issue of QT5. I thought I had properly tested this possibility earlier but apparently I was mistaken (it was close to midnight when I tested this). In the file "lc_printing.cpp" the Ansi_B case should reference "QPrinter::Tabloid" rather than "QPrinter::AnsiB". QT5 maps "AnsiB" to "Ledger", however, "Tabloid" has the same orientation as Letter/AnsiA, AnsiC,AnsiD and AnsiE.

Moreover, I did a quick test of creating and exporting a simple rectangle, as outlined above, but specifying "Ledger" paper, instead of "ANSI B". The results were similar to that of the Ansi B case. The paper was oriented in portrait mode and the rectangle in landscape mode. Perhaps, the "Ledger"  has the same issue as "Ansi B".

Item 2), ANSI B label changes to "Ledger" in the "Options" -> "Current Drawing Preferences" -> "Paper" tab -> "Format"
As both Ansi B and Ledger are the same physical size, a hack is to simply to change one of the dimensions by a small amount. I arbitrarily chose to decrease the Ansi B height by 0.1 mm. This works, but has side effects. I find this tolerable, but undesirable, a better solution is needed.

Furthermore, a similar problem occurs with "ANSI A" and "Letter". The "ANSI A" size reverts to "Letter" on re-opening the "Current Drawing Preferences" -> "Paper" tab.

Suggested solution: That a separate "ANSI A" and "ANSI B" be omitted and that the "Letter" and "ANSI A" be combined, the label be changed to "Letter/ANSI A". Also that "Ledger" and "ANSI B" be combined and the label be changed to "Ledger/ANSI B". (This works if the lc_printing.cpp line 44 entry for "Ledger" is changed to "QPrinter::Tabloid". 

If a true tabloid format is desired it can be put back in. HOWEVER, remember the QT idiosyncrasy if tabloid is brought back, it will need be connected to QT with "QPrinter::Ledger" in lc_printing.cpp!

Please note also, I'm working from QT 5.9.8 (nowebengine), it is possible that these issues are unique to this version of QT5, or to a limited subset of QT5. More evaluation of QT5 is needed, as well as cognizance of this quirk in future versions of QT5.

The following is a suggested patch to addresses all the above.

<1247.diff>

     diff -Naur a/librecad/src/lib/engine/rs.h b/librecad/src/lib/engine/rs.h
     --- a/librecad/src/lib/engine/rs.h	2020-06-11 06:09:37.000000000 -0400
     +++ b/librecad/src/lib/engine/rs.h	2020-06-13 09:06:48.909094394 -0400
     @@ -896,13 +896,12 @@
      
      
              /* US "Office" */
     -        Letter,   /* 216 x 279 mm	8.5 x 11.0 in */
     +        Letter,   /* 216 x 279 mm	8.5 x 11.0 in (Also "ANSI A") */
              Legal,    /* 216 x 356 mm	8.5 x 14.0 in */
     -        Ledger,   /* 279 x 432 mm	11.0 x 17.0 in (Also "Tabloid".  Although, technically, "Ledger" is 17 x 11.) */
     +        Ledger,   /* 279 x 432 mm	11.0 x 17.0 in (Also "ANSI B") */ 
     +            /* Also "Tabloid".  Although, technically, "Ledger" is 17 x 11.)  */
      
              /* ANSI */
     -        Ansi_A,   /* 216 x 279 mm	8.5 x 11.0 in */
     -        Ansi_B,   /* 279 x 432 mm	11.0 x 17.0 in */
              Ansi_C,   /* 432 x 559 mm	17.0 x 22.0 in */
              Ansi_D,   /* 559 x 864 mm	22.0 x 34.0 in */
              Ansi_E,   /* 864 x 1118 mm	34.0 x 44.0 in */
     diff -Naur a/librecad/src/lib/engine/rs_units.cpp b/librecad/src/lib/engine/rs_units.cpp
     --- a/librecad/src/lib/engine/rs_units.cpp	2020-06-11 06:09:37.000000000 -0400
     +++ b/librecad/src/lib/engine/rs_units.cpp	2020-06-13 09:10:23.599100143 -0400
     @@ -859,10 +859,6 @@
              return RS_Vector(279.4, 431.8);
              break;
      
     -    case RS2::Ansi_A:  /* 8.5 x 11.0 in */
     -        return RS_Vector(215.9, 279.4);
     -    case RS2::Ansi_B:  /* 11.0 x 17.0 in */
     -        return RS_Vector(279.4, 431.8);
          case RS2::Ansi_C:  /* 17.0 x 22.0 in */
              return RS_Vector(431.8, 558.8);
[... 30 lines cut ...]
     -        return RS2::Ansi_A;
     -    if (ls == QStringLiteral("ansi b") || ls == QObject::tr("ansi b", "Paper format").toLower())
     -        return RS2::Ansi_B;
[... cut ...]
          case RS2::Letter: return QPrinter::Letter;
          case RS2::Legal:  return QPrinter::Legal;
     -    case RS2::Ledger: return QPrinter::Ledger;
     +    case RS2::Ledger: return QPrinter::Tabloid;
      
     -    case RS2::Ansi_A: return QPrinter::AnsiA;
     -    case RS2::Ansi_B: return QPrinter::AnsiB;
          case RS2::Ansi_C: return QPrinter::AnsiC;
          case RS2::Ansi_D: return QPrinter::AnsiD;
          case RS2::Ansi_E: return QPrinter::AnsiE;

</1247.diff>
 
Thanks all.

slackuser0xae34, 2020-06-15:
Closed via 1249, Thanks all.

lordofbikes, 2020-06-15:
Have you build the version from #1249?
Can you confirm that it works?

slackuser0xae34, 2020-06-16:
lordofbikes,

Yes I can confirm that the patch works. I built the patch into LibreCAD-master, (9f2c32dd4b9dc2ed5e028b682473b098c48ad3e0, 2020/06/15) on both Slackware 14.2 64 bit and 32 bit. Both compile with the patched master correctly, I tested the specifics of the patch on my 32 bit machine. The changes work there. I expect the same for the 64 bit machine.

Thanks, slackuser0xae34

lordofbikes, 2020-06-16:
Many thanks for testing and validating this pull request.
The issue will be closed with the merge of #1249.

lordofbikes, 2020-12-28:
merged #1249

closed 2020-12-28 by lordofbikes