LibreCAD/LibreCAD #1248: Command line arguments for 'text' command

ch4itu, 2020-06-13:
Hi Friends!

When we invoke text command a dialog appears. Is there any method to input arguments in the command line itself? This will be useful for drawing text while psuedo-scripting.

I have gone through the plugins and issues #689 , #256 and #782 but with no avail as I am unable to write a plugin as I am not proficient with C++.

Thank you.

Chaitanya Kumar Jagabathula

### Expected behavior

### Observed behavior

### Steps to reproduce or sample file

### Operating System and LibreCAD version info

fa201, 2020-06-13:
Please have look at the User Manual to see if it helps.
https://librecad.readthedocs.io/en/latest/guides/cmdline.html

ch4itu, 2020-06-13:
> Please have look at the User Manual to see if it helps.
> https://librecad.readthedocs.io/en/latest/guides/cmdline.html

Thank you for the reply.

But the command "text", or "mtext" for that matter, opens a dialog which will not be useful for pseudo-scripting.

"text" command cannot be used in a "command file". I was looking for input in command line for all the arguments/parameters for "text" command.

Regards,
Chaitanya Kumar Jagabathula

fa201, 2020-06-13:
Sorry I misunderstood command line and text command.

c-hartmann, 2020-10-09:
Any progress on this? Any Solution to enter text without the dialog?   many thx and best regards