LibreCAD/LibreCAD #1252: LibreCAD forgets non-standard fonts

slackuser0xae34, 2020-06-23:
Problem: LibreCAD "forgets" non-standard fonts. If I install new lff fonts in a custom location, upon closing and reopening a drawing the font is "forgotten" by LibreCAD.

Actually, this is incidental to my task. I want to create a table in LibreCAD (Switch positions versus contact connections) adjacent to a switch module. The "MText" command doesn't implement the tab character to allow the alignment of multiple entries on several lines. The old trick of a mono-spaced font and counting characters is the work around. Below, came from the effort to obtain a mono-spaced font.

BTW, even using a mono-spaced font (LiberationMono-Regular, I check this in LibreOffice Writer) LibreCAD doesn't mono-space the text. Since mono-spacing doesn't appear to work, am I really using the "LiberationMono-Regular.lff" file or is LibreCAD using something else and not telling me? There may be several issues here.

### Steps to reproduce or sample file

0. Create and install a non-standard font, in a non-standard location. For example (on Linux)
	Verify that there are font files in /usr/share/fonts/TTF/
	Verify that there is a font file: /usr/share/fonts/TTF/LiberationMono-Regular.ttf. Most modern Linuxes have the Liberation fonts installed. Mac and Windows users may want to obtain them. If there is no file, use a different font file.
	As root:
	# mkdir /opt/lff_fonts	
	# ttf2lff /usr/share/fonts/TTF/LiberationMono-Regular.ttf /opt/lff_fonts/LiberationMono-Regular.lff
	Verify that file /opt/lff_fonts/LiberationMono-Regular.lff exists.
	Exit root and return to regular user mode.

1. Open LibreCAD

2. Navigate to "Options" -> "Application Preferences". Select the "Paths" tab.

3. Navigate the "Fonts:" entry to "/opt/lff_fonts"
	Either delete the default path and type in the new/above path, or
	Click the folder icon and the "Find Directory" box appears. Navigate to the "/opt" directory, .
	Single click on the "lff_fonts" folder/directory, then click the "Choose" button. (Remember, you are looking for a path to files, not the files themselves.)
	Verify that the "Fonts" line indicates "/opt/lff_fonts"

4. Click the "OK" button on the "Application Preferences" box.

5. Close LibreCAD.

6. Re-open LibreCAD (Closing and reopening is needed to re-read the path data)

7. Verify that "Options" -> "Application Preferences" -> "Paths" -> "Fonts" shows "/opt/lff_fonts".

8. Open New Drawing, <Ctrl-N> or "File" -> "New"

9. Draw some text
	"Command" box: "mtext"
	In the "Font" pull down, select "LiberationMono-Regular" (or the selected test font file).
	Set the "Height" to something reasonable, for example "0.1"
	Text in the "MText" -> "Text" box.
		Example (two lines of text, stacked vertically):
			"11111iiiiilllll"
			"00000OOOOOWWWWW"
			(The second line is five zeros, then five upper case letter O, and five upper case letter W.)
	Click "OK" on the "MText" box.
	Specify the insertion point, for example, "0,0". Otherwise, click somewhere in the drawing area.
	Press "Escape" to stop drawing text.

10. Press the "Auto Zoom" button.

11. Verify that the font is still correct
	"Command" box: "prop"
	Click on the text, and the "MText" box appears.
	Verify that the "Font" is still the selected font (in this case "LiberationMono-Regular")

12. Save the drawing.

13. Close the drawing. LiberCAD itself can be closed, or left running.

14. Open the drawing again.

15. Check what the font is
	"Command" box: "prop"
	Click on the text, and the "MText" box appears.
	Examine the "Font" pulldown.

### Expected behavior
The font pulldown should still indicate "LiberationMono-Regular". Also the length of both example lines should be the same.

The fonts installed with LibreCAD (in the default LibreCAD font directory) work correctly.

### Observed behavior
The font pulldown is blank. Also the length of lower example line is much longer than the upper example line.

### Operating System and LibreCAD version info
OS: Linux
	a) Slackware 14.2 Linux
	b) QT 5.9.8 (nowebengine)
	c) MuParser 2.2.5
LibreCAD
	a) LibreCAD-master (from Github)
	b) Identifier: 9f2c32dd4b9dc2ed5e028b682473b098c48ad3e0
	c) Date: 6/11/20, 11.22 AM
	d) The 1149 patch has been applied

### Notes
Troubleshooting included:
1. Check that the correct "ttf2lff" is used. It is, there is only the one ttf2lff in my computer, installed with LibreCAD

2. Return the "Options" -> "Application Preferences" -> "Paths" -> "Fonts" entry to the default (/opt/librecad/resources/fonts), then copy a copy of the "LiberationMono-Regular.lff" font file to the standard LibreCAD fonts directory. The result is that LibreCAD still forgets about the "LiberationMono-Regular.lff" font.

3. Make three line sets of the step 9 text. Set one set to "simplex" font, set the second set to "standard" font and set the third set to "LiberationMono-Regular" font. It appears that "LiberationMono-Regular" is identical to "standard".

4. IMPORTANT, digging into the saved DXF file, LibreCAD properly records the font as "LiberationMono-Regular", in the "AcDbTextStyleTableRecord" and the "AcDbMText" records (along with the text). The problem appears to be when the DXF file is read in to LibreCAD on opening the DXF file.

5. I've not gotten into the code yet.

slackuser0xae34, 2020-06-25:
Update 2020/06/25,

Troubleshooting the code has established that this is a problem of case sensitive OS (Linux/Mac(BSD)) versus case insensitive OS (Windows), rather than "non-standard" font file names or locations. 

WORKAROUND: The simple work around is to either rename all font files which have file names containing upper case letters to file names without upper case letters. On Linux, soft links with all lower case letters work as well.

Further, there may a potential problem on Linux & Mac (a BSD derivative) where two different fonts may have file names differing only in letter case. I haven't checked yet, but is there a possible issue with two DXF files, in the same directory, whose names differ only by letter case?

## Issue 1

Part of the problem is in: LibreCAD-master/librecad/src/lib/engine/rc_fontlist.cpp 

A file name is being compared to a filename converted to lower case. In the function, RS_FontList::requestFont(const QString& name).

The variable "name2" is a lower case version (line 97) of the desired font file name. Later (line 112) the "name2" variable is directly compared to a file name. If there is no match, then the default "standard" is used. There will be a mismatch if the font filename uses upper case letters.

Fix (partial): Change Line 112
from:
        if (f->getFileName()==name2) {

to
        if (f->getFileName().toLower()==name2) {

BTW, this issue also affect the LibreCAD default fonts "OpenGostTypeA-Regular.lff" and  "OpenGostTypeB-Regular.lff". The issue is even in the LibreCAD 2.2.0 user manual. if you look closely at the font examples on page 90 (pdf pp 94), both OpenGostType fonts appear very similar to the "standard" font. With the above change, they are noticably different from "standard" when used in LibreCAD.

## Issue 2

1) The above makes the fonts useable. However, when a upper case font is selected, LibreCAD will remember the font name (in the "MText" window, "Font" pulldown) while a drawing is open. However when the drawing is closed (both closing only the drawing and/or closing LibreCAD) then re-opened then checking the "prop" (properties) of mtext, the "Mtext" window, "Font" box is again blank.

A style name is being converted to lower case. In the function,  LibreCAD-master/librecad/src/lib/filters/rs_filterdxfrw.cpp, line 663

    sty=sty.toLower();

Commenting out this line (//   sty=sty.toLower(); ) fixes this issue. However what are the side effects? This code was intentionally put here, for some reason. What would be adversely affected by it's removal? 

## Issue 3

The mono-spacing isn't exact. With LibreOffice Writer I've tested "LiberationMono-Regular" font to 150 point (2") letters and the mono-spacing is correct and consistant. Once LibreCAD starts using "LiberationMono-Regular" font, the mono-spacing isn't exact. At the moment, I'm thinking rounding error, but haven't had time to look into this yet.

For the record and if necessary, I place all code supplied above in the public domain.

slackuser0xae34, 2020-06-26:
Update 2020/06/26

#### Item 1
    LibreCAD can handle DXF filenames differing in letter case only (simple testing).

#### Item 2
    Traditional mono-spacing text doesn't appear to be doable in LibreCAD.

    Examination of the "LiberationMono-Regular" font versus the LibreCAD "standard" font shows that both have letters of varying width. In the "standard" font the letter widths vary significantly. In the "LiberationMono-Regular" font the letter widths vary less but are not identical. 
    LibreOffice Writer presents text in the font "LiberationMono-Regular" as a mono-spaced text, as expected.
    LibreCAD presents "LiberationMono-Regular" as a fixed letter-space text, together with the varying character width makes a not mono-spaced font.
    Perhaps LibreCAD can't mono-space text. On the other hand,  LibreOffice uses the "TTF" font file format, whereas LibreCAD uses a (converted) "LFF" font file format. Perhaps there is additional information in the "TTF" file format which permits mono-spacing.

    Finally, mono-spacing of characters or alignment of words can be done as manually created arrays of individual text entities, doable but laborious.

#### Interesting trick: 
    Open one of the llf font files. Each pair of square brackets on a line is a character (for US fonts, the numbers between the brackets are the character's ASCII value in hexadecimal). The entries which follow square brackets are the position of the verticies of an individual character. Each line contains one or more pairs of coordinates (x,y) of a vertex. Multiple lines for a give multiple "strokes" of a letter. For example the letter "i" has two strokes, one for the dot and one for the vertical.

    - In LibreCAD, open a new drawing. 
    - In the "Command" box, type "pl" (for polyline). 
    - Find a character in a lff file (For example, from "standard.lff", and search for "0057" for the letter W).
    - Copy the number entries to the clipboard (<Ctrl-c>), From the example "0,9;2,0;4,6;6,0;8,9"
    - Paste the entry into the LibreCAD "Command" box, which should be asking for "Specify first point", then press <Enter>.
    - Press "Escape", to stop entering verticies for the polyline.
    - Press the "Auto Zoom" button.
    - The "W" character should appear. The dimensions are rather large (8 x 9).
    - Other characters from lff files can be treated similarly.
    - The "OpenGostType[AB]-Regular" fonts can be similarly played with. They display "hollow" letters.
    
#### Where to go from here.
    The title of this bug is incorrect, LibreCAD doesn't forget fonts. The actual issue is that LibreCAD requires font files with all lower case letters.

    I suggest that the dev's consider whether or not they wish to view the case sensitivity in font file names as a bug or characteristic. 
    If a characteristic, then I further suggest that the "OpenGostType[AB]-Regular" be renamed using only lower case letters and a note be made in the fonts section of the user manual. 
    If a bug, I've pointed to places in the code where this issue is associated. However, the issue looks to be cross-platform, and needs testing across platforms (I can't do this).

    For the user, after installation of LibreCAD, rename the ""OpenGostType[AB]-Regular" font files to all lower case letters. If the user needs to install their own fonts, then use all lower case letters in the filenames, or soft links with all lower case letters.

    As to mono-spaced text and table-like text entities there are several options.

    The most obvious is to create a single text entity, copy it to a row of entities (mv command) then copy the row to the desired number of rows (to obtain columns, again the mv command). After which change the properties of each entity to the desired content. Laborious and tedious but straight forward.

    A second mechanism is to create a table in another application then export the table to a png file. The png file can be imported into LibreCAD. 

          Application               Export to PNG
          LibreOffice Calc          YES 
          LibreOffice Writer        YES 
          LibreOffice Draw          YES 
          Apache OpenOffice Calc    NO
          Apache OpenOffice Writer  NO
          Apache OpenOffice Draw    YES
          Gnumeric                  NO
          Abiword                   NO
          Microsoft xxx             Unknown (I use Linux)
          Gimp                      YES

      Remember, normally an export to png file includes background pixels (generally white). Additional processing to remove background pixels may be desirable (ImageMagick or Gimp can do this). Simple tables can be done directly in Gimp.

    BTW, LibreCAD does recognize transparent background pixels in an imported png file. One can draw additional entities behind or in front of transparent portions of a png image.

lordofbikes, 2021-04-04:
Issue 1 was earlier fixed with https://github.com/LibreCAD/LibreCAD/commit/56dddeb332009c7772ed100e66a89f41713260d2
The issue was triggered on IRC today and I solved it before I found these excellent investigations again.
Found them now during post-treatment and applied your suggestion for issue 2 now.
Many thanks for your efforts.

Issue 1 was also duplicated in #1319 and may be related to #1243 too

lordofbikes, 2021-04-20:
follow up https://sourceforge.net/p/librecad/bugs/614/