LibreCAD/LibreCAD #1258: Properties shows wrong layer for images

dellus, 2020-07-06:
When asking Properties of an image always the active layer is indicated, instead of the actual layer. This can lead to confusion.

jbergengh, 2021-12-09:
This bug also has the undesired effect of moving the image to the current level even though you haven't changed any of the properties.

dellus, 2021-12-09:
For me the image is not really moved to the current active layer. It is still on the layer it has been inserted. You can test this by toggling it's visibility on and off. But the "Properties" box always names it's layer as the one which is currently active.

jbergengh, 2021-12-09:
That is correct.  I should have been more specific.  If I view the image's properties, but do not change any properties, and then click OK, the image gets moved to the current active layer.  Because it incorrectly defaults to the active layer instead of the image's layer.  This drove me mad until I read about this bug and learned to hit Cancel instead of OK even though I wasn't changing any properties.

dellus, 2021-12-09:
Ah, now I see. Yes, I can confirm this. So it's not only confusing but really a problem.

fa201, 2021-12-10:
Confirmed on 
Version: 2.0.7-1810-geba14138
Compiler: GNU GCC 9.3.0
Compiled on: Nov 30 2021
Qt Version: 5.12.8
Boost Version: 1.71.0
System: Ubuntu 20.04.3 LTS

closed 2023-08-01 by dxli