LibreCAD/LibreCAD #1259: Image prevents selection of line entities

dellus, 2020-07-06:
If you try to select an entity which is lying within an image by clicking on it, the image gets selected instead. If you lock the layer of the image (in the case it has an extra layer), no selection occurs either, it is blocked.
It is possible to select such entities by pulling up a window around them. But some operations need single click on selection.
It works in Draft Mode or with the image's layer switched not visible, but often the image is needed as background to position and edit line entities.
So line objects should be given selection priority over images. An image should only be selected when clicking into a no line object area or at the image's border.

V 2.1.3, 2.2.0 rc, not OS specific.

slackuser0xae34, 2020-07-07:
Might this be an issue of stacking. Is it possible that the image is stacked on top of the entity desired? If so, then selecting on an entity, on which an image is over top, would select the image. The aforesaid assumes that even transparent image pixels are selectable. Try changing the order of the stacking of the image versus entity.

Also, it appears that there at least two forms of stacking, layer stacking and entity stacking. This idea refers to entity stacking. Layer stacking would likely have nothing to do with this.

In my defense, I haven't played with this much. Thus, I'm throwing out an idea, which I haven't explored.

dellus, 2020-07-07:
If the image is stacked on top of the entity it hides the entity completely. If it is at the bottom, as it often will be the case (for instance placing a satellite or arial photo beneath a building site plan), it behaves as I described. So this is not the reason.

slackuser0xae34, 2020-07-08:
Dellus, I haven't seen this particular problem, but it interests me. I hope you will forgive me for throwing out ideas, since I can't replicate the problem myself.

Might this be a mouse/OS problem? In that when the mouse pointer is "over" an entity, the OS/GUI reports it's position to LibreCAD. LibreCAD then selects the entity. There is some "slop" in how near a mouse pointer is to an entity to be reported as "over". This position slop is set in the OS. In my case, of Linux/Fluxbox, I can set the specific value in a configuration file. Is it possible that the slop, combined with a narrow line width, may be such that you must be extremely precise in the mouse positioning? Perhaps, the numbers are such that one can't actually select the non-image entity (due to the normal fuzz in human mouse movements).

Perhaps testing with fat lines will shed some light.

dellus, 2020-07-08:
Also doesn't work with fat lines. I have attached a sample file plus image.
[imagetest.zip](https://github.com/LibreCAD/LibreCAD/files/4891810/imagetest.zip)
Tested in LC V2.1.3 on Win XP and PuppyLinux Ubuntu Beaver derivative.
Try Trim Two on the lines. For me no success. But works if Draft mode is on or the image layer made unvisible.

slackuser0xae34, 2020-07-09:
Dellus,

I was able to devote some time to this. I created a document which exhibits the bug, thus can confirm it's existence. I did some digging into the code and have some preliminary results, alas, no solution. These results are as follows:

On a mouse click near a coincident entity and image, in rs_snapper.cpp, the code gets the mouse coordinates. Calls to rs_entitycontainer.cpp iterates through a list of visible entities to see which entity is within the snap range. 

Unfortunately, when images are imported they appear to be always placed in the entity list ahead of everything else. The index position of all non-images is adjusted to accommodate the image(s) being at the head (indexes closest to 0) of the list. Similarly, second images are also placed ahead of all other entities, except prior images.

In addition, a variable for resolution is included to exclude text/images from being selected ahead of other entities. However, in this use, the "level" variable (for RS2::ResolveLevel) doesn't appear to be set to exclude images.

1st swag at identifying the explicit problem,

1) On a mouse click (release actually), the following is called (in rs_snapper.cpp):
"RS_Entity* RS_Snapper::catchEntity(QMouseEvent* e, RS2::ResolveLevel level)" 
which calls
"catchEntity(RS_Vector(graphicView->toGraphX(e->x()),graphicView->toGraphY(e->y())),level)"
	The instance used is of a RS_Vector (x,y coordinates of the mouse cursor) and the level, which is unset. Can "level"actually be set here? There is only a set of coordinates from the mouse and which entities are near by is unknown.
	
2) On being called (in rs_snapper.cpp) 
"RS_Entity* RS_Snapper::catchEntity(const RS_Vector& pos, RS2::ResolveLevel level)" 
in turn calls: 
"container->getNearestEntity(pos, &dist, level)".
	Again, just coordinates are available, dist is arbitrarily set, level is still unset.

3) On being called (in rs_entitycontainer.cpp) 
"container->getNearestEntity(pos, &dist, level)"
 in turn calls: 
"d = getDistanceToPoint(coord, &e, level, solidDist)"
	Again, just coordinates are available, the "&e" (ex-dist) is passed through, level is still unset, solidDist is arbitrarily set.

4) At line 1436 of rs_entitycontainer.cpp (comments omitted for clarity)
double RS_EntityContainer::getDistanceToPoint(const RS_Vector& coord,
                                              RS_Entity** entity,
                                              RS2::ResolveLevel level,
                                              double solidDist) const{

	double minDist = RS_MAXDOUBLE;      // minimum measured distance
 	double curDist;                     // currently measured distance
	RS_Entity* closestEntity = nullptr;    // closest entity found
	RS_Entity* subEntity = nullptr;

	for(auto e: entities){

        if (e->isVisible()) {
...
            // bug#426, need to ignore Images to find nearest intersections
            if(level==RS2::ResolveAllButTextImage && e->rtti()==RS2::EntityImage) continue;

            curDist = e->getDistanceToPoint(coord, &subEntity, level, solidDist);

            if (curDist<=minDist) {
                switch(level){
                case RS2::ResolveAll:
                case RS2::ResolveAllButTextImage:
                    closestEntity = subEntity;
                    break;
                default:
                    closestEntity = e;
                }
                minDist = curDist;
            }
        }
        }
    }

Here is where the iteration through the near by entities is done. Unfortunately, the level variable is still unset. This being the case the "if(level==...)" will always fail and the image used as one of the entities. 

For every point on an image and another entity, coincident with the image, the distance from the mouse cursor each will be the same. Therefore, the order in the list of entities will determine which is selected. With the images always at the head of the list, they get selected.

Solid fill hatches appear to work correctly. I expect that this is due to hatches having their own e->getDistanceToPoint(coord, &subEntity, level, solidDist) (rs_hatch.cpp: 737: "double RS_Hatch::getDistanceToPoint")

Perhaps Bug 426 has pertinent information (I've not looked yet). Perhaps part of the fix for bug 426 was changed/deleted on account of another issue. 

Unfortunately, currently I'm at a loss on how to fix this.

In addition: 
The choice of the name "level" for this variable is unfortunate. As it appears that the name "level" exists in three contexts. The contexts are LibreCAD debugging, LibreCAD simple testing and the "RS2::ResolveLevel" global variable.  Whether or not this common variable name is part of this or not, I don't know. When one of the developers has time, they might wish to look into this.

P. S. I'll look at your example when I get home, I get Internet through coffee shop wifi. Thanks S.

slackuser0xae34, 2020-07-10:
Dellus,

Have you seen: 
There's a similar issue for "Snap on entities": can not snap on entities within an overlapped image.

RS_Image::getNearestPointOnEntity(coord) has to return coord, if coord is within the image rect, to be able to click to select images within the image rect.

The current workaround is to hide image to snap on overlapped entities.

from: "https://sourceforge.net/p/librecad/bugs/485/"

I haven't tried it yet, I just this minute found it. The portion of code indicated above is where I'm troubleshooting. It may be the same problem. I'll try the workaround when I get home. The code to attempt fix this is looking pretty ugly.l Unfortunate, I can't find the detailed code where they presented a fix.
S

dellus, 2020-07-11:
Snap on entities works for me, tested with the imagetest.dxf. So this part of the issue seems to have been fixed meanwhile.

slackuser0xae34, 2020-07-13:
Dellus,

I agree, for me the situation is acceptable.

My results are that this issue is not a simple bug. This issue is a product of the basic design of the code. The issue can be "fixed", but the cost is a significant effort and changes across a number of files as far as I can determine.

In short, the code uses recursion to travel a tree of entities within the top "entity container" (aka drawing), some of the entities are either an individual entity or another "entity container", which itself must be evaluated for each of its components (any of which may be another "entity container, requiring further evaluation). The original author used recursion, which is rather clever. However, like all clever, it can come back to bite you.

Were I to attempt to fix this, I would move the recursion one level lower. There is data that needs recorded at the top/first instance, prior to recursing, that the lower recursed to levels don't need. This is hard to do (at least for me, I tend to avoid or narrowly limit recursion in my code, once burned twice shy).

Thus, the problem would best be addressed by the developers, a user/hacker like me isn't the proper place to make these sorts of changes. Further the changes would need coordination among code modules, detailed test of the code, then testing across platforms. In short, I could theoretically do it, the changes would likely be viewed with significant scepticism by the devs.

As the 485 bug has shown a work around, which by my testing works. The only thing I can foresee that I can do further is to urge the people who write the user manual to note this issue and the work around boldly in the manual portion assigned to images.

On the other hand, perhaps someone with more skill/experience can develope a simple solution, which I don't see.

Thanks for the code fun. S.

natrius, 2021-04-23:
The issue is still present in 2.2.0 rc-2 on Windows 10. If the layer with the imported image is locked, it does not select the image, but the line not as well. If i hide the image-layer, i'm able to select the line.

In my opinion it should just be possible to select a image when clicking on the border, not withing the image itself.

kmpm, 2024-01-24:
It works in qcad. Is there any licence compatible way one could look at how they have solved it?

lordofbikes, 2024-01-24:
LibreCAD was forked nearly 2 decades ago from QCAD version 1. Any comparison with current QCAD version 3 doesn't make much sense.