LibreCAD/LibreCAD #1262: Blocks should not preserve layer property

Ceaus, 2020-07-12:
When you save a block to a file, it preserves the layer associated with that block. 
In consider this a bug, as layers are an aspect of organization (meta level). They are not part of the element itself. It's really a different dimension.
This gets especially annoying if you import someone else's blocks. Then you get to have their layers too, messing with *your* organization.

xanderdin, 2020-07-12:
I would disagree because a block can contain elements from many layers. They shouldn't be moved blindly to a single layer on block saving because each layer may have different properties. It could be convenient if layers named properly, and then importing such saved block into another drawing you can see what layers were imported from which block drawing.

Ceaus, 2020-07-12:
I totally follow your reasoning and logic. Still I disagree on a rather fundamental level. 

1. A layer is and remains an organizational/meta feature. With that it becomes a kind of _separation of concerns_ issue. You are basically allowing someones brain products to enter your design. While you are really only want the blob. The choices the block creator has made to organize his stuff could be totally irrelevant to your design. And now you need to align his meta choices with yours: colors, widths, etc, etc. Because whichever way you look at it, it is messing with your design. So in reality you've just tossed the decision of the block creator out of the window. Which is more or less the outcome of this GitHub issue.

2. Blocks are by definition isolated/standalone (can't find a good word for it) entities. In the sense that their life is meaningless until you've incorporated them into your design. I would therefor argue that if you need layers to organize a block, you've made the block too complicated. The complexity should be within your design. Because that's where the magic happens. The block is just a block.

dellus, 2020-07-13:
I think we should stick with AutoCAD standards in this question. In AutoCAD layer information is saved during block creation and when inserting the block into a drawing these layers are created if not existing. So it's up to the block creator to make a simple one layer block. If you have the original entities on layer 0, later the block will be inserted into the then active layer.
If you want to make a one layer block out of multi layer entities, a procedure could be to copy these to a free space on the screen, select them and shift them to layer 0 with Modify - Attributes, then create the block.

dellus, 2020-07-13:
Maybe an additional alternative block insertion mode could be introduced, with dumping all the original layer information.

Ceaus, 2020-07-13:
> Maybe an additional alternative block insertion mode could be introduced, with dumping all the original layer information.

I think that would be the most comfortable approach.

Ceaus, 2020-07-13:
> I think we should stick with AutoCAD standards in this question. In AutoCAD layer information is saved during block creation and when inserting the block into a drawing these layers are created if not existing.

I just did a quick Google on " Autocad blocks layers" and that is a world of hurt. So following the defacto standard is not necessarily beneficial. 

Also, I'd really like to hear from someone a real use case. A situation were the layers of the creator were useful in the project without modification. My guess is that 99% of the cases the layers are going to be modified in order to fit the block properly in your own project.

slackuser0xae34, 2020-07-13:
This topic appears to generate strong opinions. Please let me suggest an idea, which I think should be discussed. The ideas would allow for both sides of this, as well as address other issues I've seen but not raised.

First, the idea is predicated on the axiom that the DXF file format allows for one or more "comment" field/sections. I haven't been able to find a full DXF specification to see if there is a comment field/section. If there is not such an entity in the DXF file format, then please disregard the following.

I suggest that LibreCAD develop a specification for a set of parameters/entities/etc pertinent and useful to LibreCAD, which are stored in the (hoped for) comment field/section. Such LibreCAD specific material would be ignored by AutoCAD. Of course AutoCAD would not implement anything in the comment field.

On the other hand, LibreCAD has the option to develop features that AutoCAD doesn't have. Further, it may be easier to implement some features using a comment field to store data. This issue is an example. A standard, AutoCAD compatible block is stored as it is now. The alternative sort of block can be stored in the comment field.

The aforesaid is a proposed idea, pertinent to this discussion. I suggest that all interested may wish to think about the issue. The devs would have extra work implementing this idea, thus a big voice in the discussion. Still, the idea of LibreCAD specific items in a comment field may be useful. 

Thanks S.

lordofbikes, 2020-07-21:
@slackuser0xae34 we had this discussion already years ago. By common consent we cancelled this approach, it is too vulnerable.
The discussion arose with the construction layer feature, which was or maybe still is a special LibreCAD feature. It was stored in the layer print flag in past and always a source of issues back then. Now it is stored in DXF conform application data, what is what you described, but more secure than abusing comment sections.
I can't see how this can help for this issue.

lordofbikes, 2020-07-21:
@Ceaus I downloaded a foreign text document, formatted in 24pt Comic Sans. When I copy/paste parts to my own 11pt Arial document, the foreign text is still 24pt Comic Sans. Is this a bug in my preferred word processor?

Seriously, nor can I find a document that approve your position on layers and blocks, nor can I agree with your argumentation.

Blocks can be simple or complex, this is not a matter of too complicated.
Even small blocks can benefit from using multiple layers. I know many use cases where this is very useful.
Maybe this is only a matter of personal taste at least, but there is no accounting for taste.

How often does this occur? Can you mention a value in hours per month or so?
Then we can estimate whether the benefit is worth the effort to implement this.

Your issue is, that you have grabbed some foreign block library which does not match your personal preferences.
So the options are not to use the foreign library and create the blocks yourself.
Or find another library which fulfills your preferences.
Or to appreciate the efforts somebody put into the library and add little of your own effort to adjust its appearance.
Don't forget to check the library license and to share the modified library to give something back to the community.

Ceaus, 2020-07-22:
> @Ceaus I downloaded a foreign text document, formatted in 24pt Comic Sans. When I copy/paste parts to my own 11pt Arial document, the foreign text is still 24pt Comic Sans. Is this a bug in my preferred word processor?

That is a solid analogy. You are correct, it cannot be a bug in LibreCAD. I will remove that wording from this issue.

There is a small disconnect with this analogy though. But for the "bug" perspective it's not relevant:
If you copy stuff from someone else's document, you are copying that from a whole. With everything in it. My view of importing external blocks is more aligned with the view of importing a code snippet. But maybe my view is wrong.

> Blocks can be simple or complex, this is not a matter of too complicated.
> Even small blocks can benefit from using multiple layers. I know many use cases where this is very useful.
> Maybe this is only a matter of personal taste at least, but there is no accounting for taste.

Yes, yes, yes.

But the question is not if simple blocks can/need layers. That is obviously very clear. And I totally agree they can.
The question is if blocks, which by definition are meant to be imported into someone's design, should have their layer modeling retained after import. Maybe I'm overthinking this, but to me this seems really odd. I'd settle for an option to have all layers squashed into one import layer, with all settings set to default.

Considering the amount of force back I'm receiving it seems this issue is not resolving a real life problem (except my personal frustration). So by now I'm ready to see this issue rejected.

> How often does this occur? Can you mention a value in hours per month or so?

I prefer to measure this in frustration/annoyance rather than time. For *every* block I'm importing I need to incorporate her/his layers organization into my design.

CADbloke, 2020-07-24:
This is non-standard behaviour. It belongs in a plugin / extension. I see your use-case but it breaks accepted practice. Perhaps you could pre-process a block library with a standards conformity tool?

DuncanLithgow, 2021-04-03:
Sounds like this discussion has run its course.

closed 2021-04-03 by DuncanLithgow