LibreCAD/LibreCAD #1263: Does anyone know how to convert the hex data corresponding to group code 310 into image?

JasonStepan, 2020-07-14:
SECTION
  2
ENTITIES
  0
OLE2FRAME
  5
281
330
1F
100
AcDbEntity
  8
0
100
AcDbOle2Frame
 70
     2
  3
Picture (Device Independent Bitmap)
 10
2147.279748970612
 20
2248.34104767737
 30
0.0
 11
4530.180851497799
 21
633.0270850613914
 31
0.0
 71
     3
 72
     0
 73
     2
 90
   816256
310
8055BCCF413B8FC6A0405C19CD9DAE90A14000000000000000007CA4484C2EB2
310
B1405C19CD9DAE90A14000000000000000007CA4484C2EB2B14018675F7837C8
310
83400000000000000000BCCF413B8FC6A04018675F7837C88340000000000000
310
000051383C260100000000010000020000000000000000000100000000740C00
310
D0CF11E0A1B11AE1000000000000000000000000000000003E000300FEFF0900
310
0600000000000000000000000D00000001000000000000000010000002000000
310
01000000FEFFFFFF000000000000000004000000050000000600000007000000
310
08000000090000000A0000000B0000000C0000000D0000000E0000000F000000
310
FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
310
FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
310
FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
310
FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
310
FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
310
FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
310
FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF

 **In fact, the dxf file contains a lot of hex data with a code of 310, I just listed a few.I thought these hex data were generated in png format or jpg format, but I used the online tool HEX TO IMAGE, and these hex data could not be converted into pictures.**

lordofbikes, 2020-07-14:
With pleasure I asked [Google](https://www.google.com/search?client=firefox-b-d&sxsrf=ALeKk0252wmEsydBU2IOlHYsguzsHSuwbQ%3A1594724840429&ei=6JENX8rfGcj4kwWnwoXQCA&q=OLE2FRAME+310&oq=OLE2FRAME+310&gs_lcp=CgZwc3ktYWIQAzIHCCEQChCgAVCDjhtY4ZAbYKKeG2gCcAB4AIABngGIAckDkgEDMC4zmAEAoAEBqgEHZ3dzLXdpeg&sclient=psy-ab&ved=0ahUKEwjK84L9zMzqAhVI_KQKHSdhAYoQ4dUDCAs&uact=5) for you and here are the most valuable results:
http://docs.autodesk.com/ACD/2011/ENU/filesDXF/WS1a9193826455f5ff18cb41610ec0a2e719-79f4.htm
https://adndevblog.typepad.com/autocad/2013/02/dxf-files-binary-chunk-interpretation.html

In short, the binary data represents a BITMAPINFO structure. Presumably this is a Microsoft/Windows structure you have to look for.

closed 2020-07-14 by lordofbikes

JasonStepan, 2020-07-15:
> Microsoft

> With pleasure I asked [...]

Thank you.