LibreCAD/LibreCAD #1264: Chinese display error problem, parsing DXF error

qduck, 2020-07-16:
[image]

lordofbikes, 2020-07-16:
What do you think is our template for?

### Steps to reproduce or sample file
???
### Operating System and LibreCAD version info
???

This looks like an already solved issue, have you tried latest builds?

qduck, 2020-07-27:
> What do you think is our template for?
> 
> ### Steps to reproduce or sample file
> ???
> 
> ### Operating System and LibreCAD version info
> ???
> 
> This looks like an already solved issue, have you tried latest builds?

[image]
when i open the dfx file.

lordofbikes, 2020-07-27:
You should try a newer software first, as I assume, that this issue is already solved
https://sourceforge.net/projects/librecad/files/Windows/NightlyBuilds/master/

DuncanLithgow, 2021-04-03:
hi @qduck - did the newer version solve this issue for you? If you're running Windows then @lordofbikes link will help. Otherwise you can ask on the forum if you need help compiling a new version for your platform.

closed 2026-10-07 by dxli