LibreCAD/LibreCAD #1266: What is current version???

atorrey, 2020-07-28:
I am having a lot of trouble w/ my LibreCAD install - it is constantly hanging and going into a non-responding mode especially if I try to zoom in or pan a drawing so that part of it is outside the active window.

When I do 'about' in the program, I get:

Version: 2.1.3
Compiler: GNU GCC 8.3.0
Compiled on: May 16 2019
Qt Version: 5.11.3
Boost Version: 1.67.0
System: Debian GNU/Linux 10 (buster)

I just went through a good bit of trouble figuring out how to add the Ubuntu PPA to my sources.list in order to get the latest version since I didn't want to waste time trying to fix a problem in an out of date version.   Synaptic says the latest / installed version is 2.1.3-1.2

Is this current?   I can't find anyplace on the web-page or elsewhere that says what the current version is (Suggestion to add it???)

If it isn't, what is the current version and how do I go about getting it installed on Debian 10.4 (Buster)

If it is current, what do you suggest to either make it stop hanging or provide more useful info for bug fixing?

Thanks,
ART

lordofbikes, 2020-07-28:
> what do you suggest to either make it stop hanging or provide more useful info for bug fixing?

Not deleting the issue template, but fill in the requested info, especially **Steps to reproduce or sample file**

Your version is all you can get as binary distribution for Linux yet.
You can try to build from source, this is not so tough on Linux, see https://librecad.readthedocs.io/en/latest/appx/build.html

There are a couple of issues solved in latest code, especially zooming with dashed construction lines.
One is still open, zooming with splines.
Most are mentioned in #915.

atorrey, 2020-07-28:
OK, I may take a look at trying to compile it.  I am having the problem with most of the drawings I've done, but while they aren't terribly complex, one thing that I almost always use is dashed or dotted construction lines....

ART

------------------
Arthur Torrey - <arthur_torrey@comcast.net>
-------------------

WebDust21, 2021-01-05:
Having a similar problem...related again to dotted and dashed construction lines causing an instant, 100% reproducible terminal lockup.  Latest available on Ubuntu's Software Center is 2.1.2-1, updated 08/08/2020.
Tried the repository linked on the wiki.  You guessed it: "404  Not Found [IP: 91.189.95.85 80]".

What's confusing me is that the actual LibreCad program indicates the following version information:

Version: 2.1.2
Compiler: GNU GCC 6.2.0
**Compiled on: Sep 17 2016**
Qt Version: 5.9.5
Boost Version: 1.62.0
System: Ubuntu 18.04.5 LTS

Compiled 5 calendar years ago???  What in the world is going on?
At least for now, I guess it's no dashed/dotted lines on layers.  Makes it awfully difficult to figure out which of the 5 layers is hiding the main design though...

lordofbikes, 2021-01-24:
To put it straight, this is about missing PPA maintainer, which is tracked in #894

closed 2021-01-24 by lordofbikes