LibreCAD/LibreCAD #1268: Duplicate overwrite warning in print to file (PDF)

tochimoreno, 2020-07-31:
### Observed behavior
It launch two "File already exists" warning

### Expected behavior
Only one warning is enogh

### Steps to reproduce or sample file
1. Print
2. Select Print To File (PDF)
3. Clik "..." to select Output File (to a file that already exist)
4. Click "Save"
5. Launch Overwrite warning message
6. Click "Yes"
7. Click "Print"
8. Launch second Overwrite warning message
9. ...

### Operating System and LibreCAD version info
Version: 2.2.0-rc1-254-g9f2c32dd
Compiler: GNU GCC 9.3.0
Compiled on: Jun 11 2020
Qt Version: 5.12.8
Boost Version: 1.71.0
System: Ubuntu 20.04.1 LTS

fa201, 2020-07-31:
Strange, I do not have the same behavior. I have the first overwrite warning (which I expect) but I do not have the second one.
The second one is: 
_Paper size and/or margins have been changed!_ 
_Do you want to apply changes to current drawing?_

Version: 2.2.0-rc1-254-g9f2c32dd
Compiler: GNU GCC 7.5.0
Compiled on: Jun 15 2020
Qt Version: 5.9.5
Boost Version: 1.65.1
System: Ubuntu 18.04.4 LTS

tochimoreno, 2020-07-31:
The first warning has the librecad logo as a windows icon, "Print to file ..." title, an yellow exclamation mark as a image, and say:
<files_name_without path.pdf> already exists.
Di you want to replace it?
YES NO

This appear when I select a file en the file select dialog.

The second one is similar but "Print" as title, green question mark and include the full path in the file name.
It appears when I click Print in the "Print" dialog.

They are not the warning about de margins witch sometime appears when it has to.

May be one of then are triggered by the PDF printer and not by LibreCAD?

DuncanLithgow, 2021-04-03:
I'm going to guess that one of the notices is coming from the PDF printer. Please re-open this bug if you still think this is a LibreCAD problem.

closed 2021-04-03 by DuncanLithgow