LibreCAD/LibreCAD #1270: File does not print into pdf, just produces empty page

rdorsch, 2020-08-03:
I converted a dwg drawing into dxf using onlineconvertfree. The dxf file opens well in librecad and I can do all the editing work, I want to do.

If I try to create a PDF file with librecadt, but I obtain only an empty page. No matter if I print to file, export to pdf, run a print preview before or not. The print preview works well.

### Expected behavior

Drawing in PDF format

### Observed behavior

Completely empty page, but the file seems not to be empty, it has a size of 2.5 MB and is approx. 10x bigger than the dxf file.

The --debug=6 output looks ok for me 

RS_EntityContainer::optimizeContours
RS_EntityContainer::optimizeContours: OK
QG_DialogFactory::requestOptions
QG_DialogFactory::requestOptions: OK
QC_ApplicationWindow::slotFilePrint(PDF)
QG_GraphicView::resizeEvent begin
QG_GraphicView::resizeEvent end
QC_ApplicationWindow::slotFilePrint: resolution is 1200
RS_EventHandler::~RS_EventHandler
RS_EventHandler::~RS_EventHandler: Deleting all actions..
RS_EventHandler::~RS_EventHandler: Deleting all actions..: OK
RS_EventHandler::~RS_EventHandler: OK
QG_DialogFactory::requestOptions
QG_DialogFactory::requestOptions: OK
QC_ApplicationWindow::KeyPressEvent: IGNORED
QG_ActionHandler::keycode()
QC_ApplicationWindow::KeyPressEvent: IGNORED
QC_ApplicationWindow::slotWindowActivated begin
QC_ApplicationWindow::slotWindowActivated begin
QG_GraphicView::resizeEvent begin
QG_GraphicView::resizeEvent end

### Steps to reproduce or sample file

The .dxf file and the output pdf I uploaded here

 http://bokomoko.de/~rd/librecad

### Operating System and LibreCAD version info

Debian Linux 10 / LibreCAD 2.1.3

rdorsch, 2020-08-04:
There is an excellent analysis of the issue here:

https://forum.librecad.org/How-to-Print-Part-of-a-Drawing-tp5710934p5719394.html

To me that seems to be a bug with an existing workaround.

lordofbikes, 2022-11-29:
This issue is caused by bottom left coordinates around 32511510,5375220.
Moving the drawing near to origin at 0,0 results in valid output.

Anyhow, with a valid paper size and factor, this should print from the original place too.