LibreCAD/LibreCAD #1272: Export as image don't show black lines

gitrobertofl, 2020-08-19:
Hi!

I have a problem with export as image:

Version 2.1.3 in Windows 20 2004 x64 on HP elitedesk 8000

DXF2007 file with several layers. Some of them has black/white color and others have other colours.
When I export as image (same issue in all formats).
Resolution: auto
Background:  white
Coloured checked.
It doesn't export black lines (gray ok).

With Background black it shows all colours (like I drew it).
With background black and coloured black/white export result is a complete black drawing
with background white and coloured black/white export all grays like black, but all lines.

Best regards

### Expected behavior

### Observed behavior

### Steps to reproduce or sample file

### Operating System and LibreCAD version info

lordofbikes, 2020-08-19:
This was improved recently (#1232) and should work with latest version.
https://sourceforge.net/projects/librecad/files/Windows/NightlyBuilds/master/

DuncanLithgow, 2021-04-03:
@gitrobertofl please reopen if this has not been solved.

closed 2021-04-03 by DuncanLithgow