LibreCAD/LibreCAD #1275: Uninstaller

Andrey5045, 2020-09-04:
Dear sirs, I just installed the latest version of LibreCAD; I wanted to install it into "d:\progs\LibreCAD", but I made a mistake, so it was installed into "d:\progs\" directory where other directories are with some other programs. I decided to uninstall and then re-install LibreCAD to a new location. I ran "d:]progs\Uninstall.exe" and it started deleting ALL the files which were in "d:\progs\".

ret394, 2020-09-28:
You're absolutely right it is. It should not touch any other files. Please wait while someone fixes it. Thanks for trying it out, you're awesome.

An extra note - I'm not associated with this project, so please don't damage my reputation by falsely stating that I am a project developer or contributor.

thanks. have a nice day. keep up the great work. 👏

[ret394](https://github.com/ret394)

lordofbikes, 2020-10-27:
Confirmed by [sourceforge ticket #610](https://sourceforge.net/p/librecad/bugs/610/)