LibreCAD/LibreCAD #1277: Editing a filesystem copy of design changes original

Kniebe, 2020-09-10:
### Expected behavior
I expect that a filesystem copy of a drawing is contained.

### Observed behavior
When editing the copy after a save the original is also changed with the same changes.

### Steps to reproduce or sample file
Create copy of drawing on the filesystem
Edit the copy and save
Check original.

### Operating System and LibreCAD version info
Ubuntu 20.04
Librecad 2.2.0

lordofbikes, 2020-09-11:
How should LibreCAD know about the copy and change the original file?
I can't reproduce this, it works as expected for me.
How did you make the copy?
Could it be a link instead of a copy? This would explain that behavior.
Please post the copy command and the output of `ls -l` in the folder of the origin and copied file.

DuncanLithgow, 2021-03-16:
please reopen this bug report if you're still seeing this issue. For now it is not reproducible.

closed 2021-03-16 by DuncanLithgow