LibreCAD/LibreCAD #1278: LibreCAD theme conflicting

Bollos00, 2020-09-15:
I have been using LibreCAD since the last semester for school projects and it has been being an excellent tool for me.
I used to use LibreCAD on Ubuntu, but two months ago I switched to Manjaro and LibreCAD theme seems to be weird.

This is how LibreCAD looks for me, I thought that it was just an adaptation to my gtk theme, but then I figured out that even when I switch to a gtk light theme, LibreCAD stayed still.

[image]

I have done a record showing this (Sorry for the bad quality). As it can be seen, every application changes the theme, but LibreCAD. What I really found weird is that the default theme is dark. That usually would not be a problem, but that way I cannot see the icons clearly.

[image]

##Informations about the system and LibreCAD:

```
Version: 2.1.3
Compiler: GNU GCC 10.1.0
Compiled on: Jul  7 2020
Qt Version: 5.15.0
Boost Version: 1.72.0
System: Manjaro Linux
```

```
OS: Manjaro Linux x86_64 
Kernel: 5.8.6-1-MANJARO 
DE: GNOME 3.36.
```

LibreCAD was installed with `pacman`.

Thanks in advance.

Bollos00, 2020-10-10:
I have just found the issue, I will post the solution here for someone whom faces the same problem or for me of the future if I ever have this issue again.

There is this environment variable `QT_STYLE_OVERRIDE` that was defined with _kvantum_, this used to force all my Qt Applications to have this theme. It will be defined in one of those files:
`~/.bash_profile`
`~/.profile`
`/etc/profile`
`/etc/environment`

For me it was located on `/etc/environment`. Once Identified, delete or comment the line that define that variable (if it is in the `/etc` directory you will need root privileges). 

Then reboot your system.

Next install the package [qt5ct](https://www.archlinux.org/packages/community/x86_64/qt5ct/).
```bash
sudo pacman -S qt5ct
```

Now open the Qt Configuration Tool (qt5ct) and choose the theme that you want.

Reference:
https://bbs.archlinux.org/viewtopic.php?id=214147&p=3
https://wiki.manjaro.org/index.php?title=Set_all_Qt_app%27s_to_use_GTK%2B_font_%26_theme_settings
https://wiki.archlinux.org/index.php/Uniform_look_for_Qt_and_GTK_applications

deric, 2020-10-27:
Looks like duplicate of #1144

Bollos00, 2020-11-09:
> Looks like duplicate of #1144

Yes. I did not see it.

Anyway, now that it is solved the issue can be closed.

closed 2020-11-09 by Bollos00