LibreCAD/LibreCAD #1279: Command Compatibility with Autocad and Draftsight need to be ensured for larger adoption of this software.

MyPersonalGitAC, 2020-09-16:
### Expected behavior

### Observed behavior

### Steps to reproduce or sample file

### Operating System and LibreCAD version info

zoom, pan, trim, extend, erase, needs to be standard with other commercial software on the market.
[90404089.pdf](https://github.com/LibreCAD/LibreCAD/files/5229117/90404089.pdf)

cshunt, 2020-09-21:
I like the project a lot but I do agree. As a command line cad operator the only software i have found that is extremely close to autocad is bricscad. This surprises me with opensource software but users are coming from different experiences. 

It also surprises me that the command line is not the first place that an opensource software implements a feature. i.e. older autocad software started with only a command line list of commands. The buttons were then added. I would assume this method lends itself to have a "module" for each command and as more commands are developed they can easily be added. The bulkyness of autocad has kept me searching for light weight alternative but many of my command line habits are hard to break with slow buttons.

Thanks for the hard work on some awsome software. I look forward to working with it.

abubkurian, 2020-10-28:
Yeah I think commands should be a bit more optimized. I don't think they need to be similar to autocad, But something like delete function should be accessible easily without typing 3 letters. Also "space" should be the default button to activate a command, the 2.2 version brings it as an option, but doesn't work well with all commands. Also the command line should be ready to receive without manually activating the cursor each time. Said that, If you stick to UI, I think it is better than autocad, after some customization of course.. :)

gerardbm, 2021-01-30:
+1! And typing a command should be easy. For example, QCAD allows you to put the cursor in the command line with only one key press of the spacebar. So you can literally press spacebar and start typing the command. AutoCAD allows you to type a command from everywhere only typing, and new versions have an autocompleter, showing the available commands, like a fuzzy-finder.

abubkurian, 2021-01-30:
> +1! And typing a command should be easy. For example, QCAD allows you to put the cursor in the command line with only one key press of the spacebar. So you can literally press spacebar and start [...]

Librecad also has this feature were you press spacebar to go to command line, but works only sometimes..

fa201, 2021-01-30:
In case it is a lack of information, please note that you can enter the command line with the following:
https://librecad.readthedocs.io/en/latest/guides/cmdline.html#the-command-line

Note also that you need to enable the SPACE BAR validation in the **Application Preferences** menu.

[image]

dxli, 2024-05-18:
We probably don't want to simulate other softwares.

closed 2024-05-18 by dxli