LibreCAD/LibreCAD #1280: LibreCAD DXF cleanup to PTC CREO import

kjfull, 2020-09-23:
[image]

I use PTC Creo to draw sheetmetal parts for laser cutting and when I do more complicated signs I import a dxf into a sketch, then create a planar feature and project the sketch as a cutout in the planar feature. This helps to simplify the sketch by using polylines instead of the lines in the dxf. Sometimes the DXF files are too complex to let me import them as a sketch into CREO, so I downloaded LibreCAD hoping I could convert lines to polylines helping the import into creo. I converted the jpg attached with a free online converter to dxf then loaded it into libreCAD and used the "create polylines from existing segments" tool. It seemed like it was reducing the amount of points but it still failed when importing into CREO. I'm wondering if there is a better way to simplify the DXF down through LibreCAD or if anyone has any recommendations on how to improve my process. It has been working with less complex drawings but when the pictures get more complicated I have to import the dxf in pieces and that hasn't even been working with this one. Any help is appreciated.

Thanks,

### Expected behavior
Simplifying lines in DXF to polylines
### Observed behavior
Nothing Changes when using the, "create polylines from existing segments" tool
### Steps to reproduce or sample file
Used Convertio.com to convert  .jpg to .dxf
### Operating System and LibreCAD version info
Windows 10, LibreCAD verson 2.1.3

DuncanLithgow, 2020-12-28:
Since what you're really doing is graphics work rather than 2D CAD detailing I suggest you look into using Inkscape. It exports to DXF and has good raster to vector conversion tools.

lordofbikes, 2020-12-30:
What comes out from convertio.co is a very rudimentary DXF in the old R12 (pre 1990) format.
But all lines are already polylines, that's the reason why nothing happens when you use the polyline tool.
The magic happens when you save the file in LibreCAD, then polylines are saved as newer lwpolylines types in DXF 2007 format.
There is nothing else you can change with LibreCAD and nothing you are doing wrong.
You should reach out to PTC Creo to find out why simple DXF files work but not complex ones.

closed 2020-12-30 by lordofbikes