LibreCAD/LibreCAD #1282: Impossible Condition

maddin200, 2020-09-28:
[librecad/src/ui/forms/qg_polylineequidistantoptions.cpp line 116

void QG_PolylineEquidistantOptions::updateNumber() {
	if (action) {
		const QString& l = ui->leNumber->text();
		bool ok;
		unsigned i=static_cast<unsigned>(RS_Math::eval(l,&ok)+0.5);
		if (!ok || (i==0 && i>100))   // <-- impossible
			i=1;
		ui->leNumber->setText(QString::number(i));
		action->setNumber(i);
	}
}

DuncanLithgow, 2021-03-16:
Could you maybe explain a bit what this bug report is about?

maddin200, 2021-03-16:
Hi, 
The variable i can't be  == 0 and at the same time > 100  ( (i == 0) && (i >100) ); might be logical or  required

fa201, 2021-03-16:
@lordofbikes  Can you  check if this condition correction can be implemented in next release ? Thanks.

closed 2021-03-16 by lordofbikes

lordofbikes, 2021-03-16:
I wonder, how you found this, it should have been detected by static analysis.
Many thanks and great praise.
For curiosity's sake I looked for the source of this issue and I found it back in the first commit 2012, seems to be original QCAD code.

commit 8b2b62d1 (2021-03-16): fixed #1282, impossible condition