LibreCAD/LibreCAD #1284: Copy & paste block with angle, duplicate the angle

tochimoreno, 2020-09-29:
### Expected behavior
Copy & Peste a block from one drawing to another, copy the block with its attributes, scales, angle, rows and columns and spacing.

### Observed behavior
It dupicates the angle in the paste drawing.

### Steps to reproduce or sample file
1.- Draw a rectangle in drawing "A", convert to block, rotate 30°
2.- Create a new drawing, "B".
3.- In Drawing "A", select block, hot ctrl-C to copy and select reference point.
4.- In drawing "B", hit ctrl-V to paste and select reference point.
5.- The block is paste, the angle in properties tool is 30°, but is seen as 60°

### Operating System and LibreCAD version info

tochimoreno, 2020-09-29:
Version: 2.2.0-rc1-254-g9f2c32dd
Compiler: GNU GCC 9.3.0
Compiled on: Jun 11 2020
Qt Version: 5.12.8
Boost Version: 1.71.0
System: Ubuntu 20.04.1 LTS

Aditionaly, all the geometrys are reset, rows, columns, spacing...

DuncanLithgow, 2021-01-29:
Confirmed

dxli, 2023-06-04:
the same issue as #1616