LibreCAD/LibreCAD #1287: Diagonal line artefact across screen

maxcel2, 2020-10-31:
### Expected behavior
A vertical line through the origin on a construction layer should disappear when the origin is moved off screen left.

### Observed behavior
A vertical line through the origin on a construction layer appears diagonally across screen when the origin is moved off screen left.

### Steps to reproduce or sample file
Open a new drawing.
Set layer 0 as a construction layer
Select line vertical, any length positive or negative and place on origin 0,0.
Move horizontal slider to right until the origin moves off screen left.
The diagonal line appears starting somewhere up the left hand Y axis depending on how far the origin is moved left and always ends in the top right hand corner.

### Operating System and LibreCAD version info
Release Linux Mint 20 Ulyana 64-bit
Kernel Linux 5.4.0-48-generic x86_64
MATE 1.24.0
Version: 2.1.3
Compiler: GNU GCC 9.3.0
Compiled on: Mar 23 2020
Qt Version: 5.12.8
Boost Version: 1.71.0

fa201, 2020-11-01:
I can reproduce the bug on Ubuntu with release 2.1.3 but it seems to be fixed on last built :
Version: 2.2.0-rc1-254-g9f2c32dd
Compiler: GNU GCC 7.5.0
Compiled on: Jun 12 2020
Qt Version: 5.9.5
Boost Version: 1.65.1
System: Ubuntu 18.04.5 LTS

maxcel2, 2021-01-29:
I've tried installing the PPA for Linux Mint 20 but get the error message:

''This PPA does not support focal''

It seems that later versions of librecad are not compatible with later versions of Ubuntu/Linux Mint

fa201, 2021-01-29:
@maxcel2, the work around is building from source code : https://librecad.readthedocs.io/en/latest/appx/build.html#building-on-linux. It is not so complicated provided that all packages are installed before compiling.

DuncanLithgow, 2021-03-16:
@maxcel2 please ask on the forum if you have trouble building the software from source. Encourage your distribution to make a package. See bug #894

closed 2021-03-16 by DuncanLithgow