LibreCAD/LibreCAD #1288: tangential two circles, point circle drawing

orangecrow, 2020-11-01:
### Expected behavior
Using tan,tan,point on a line and a circle and the point on the line should produce a drawing such that the resulting circle should be tangent to both the line and the circle and touch the line at the point specified on it.

### Observed behavior
The resulting circle goes through the specified point and is tangent to the circle but it is not tangent to the line. In fact it is crossing the line perpendicularly.

### Steps to reproduce or sample file
Draw a circle then a line roughly perpendicular to the circle surface. than use the "tangential two circles, point" option, select the circle, the line, and the point in the middle of the line.

### Operating System and LibreCAD version info
**System:**
Windows 10
**LibreCAD version info:**
Version: 2.1.3
Compiler: GNU GCC 4.9.1
Compiled on: Sep 23 2016
Qt Version: 5.4.1
Boost Version: 1.60.0
System: Windows 8
### Second  Operating System and LibreCAD version info
**System:**
Linux Ubuntu (running on a choromebook thorugh chroot)
**LibreCAD version info:**
Version: master
SCM Revision: 2.0.9
Compiler: GNU GCC 5.3.1
Qt Version: 5.5.1
Compiled on: Mar 24 2016

DuncanLithgow, 2021-03-16:
Please provide a file or sketch. I can't see how what you want is possible. If it is tangential to the circle and the line what would define the radius of such a circle?

Also you might as well install 2.2rc2

lordofbikes, 2021-03-16:
The steps are clear, no need for a sample file.
And also the issue is clear.
What I can't say yet, if this behavior is a limitation of the equation solver
or if this special case, when the point is on the line, is intended by design.
My assumption is, that both is true. For any point on the line we have two solutions, how can the tool decide, on which side of the line the tangent circle should be.
For points apart the line, the solution is clear.

jbergengh, 2021-11-29:
When there are two solutions, as in this case, user should be allowed to choose which solution is desired.  That is how the "Tangent, 2 points" tool handles it.  For example, draw a circle, select the "Tangent, 2 points" tool.  Select two points outside the circle.  There will be two solutions.  LibreCAD shows the centers of both circles and allows the user to select which solution to use.  This issue is the same situation.

closed 2023-06-06 by dxli