LibreCAD/LibreCAD #1289: ppa issue(s) : 404 not found

eglantinepourpre, 2020-11-10:
Hi, 

My boyfriend and I are working on our plans for our house and I was happy to find LibreCAD to draw it under my **Ubuntu 18.04 64 bits **. It's been more than ten years I've totally given up with windows and been working on opensource software, and your project is awesome ! Thank you for that :+1: 

We're experiencing some troubles about the PPA and, as none of us is an expert in programming or else, we'd need some help to make it work.

I've added this  ppa:librecad-dev/librecad-stable
but when we try to update it, we have a message telling us  that "the ppa does not have a release file". 

We would also be glad to have the french version of the program, we've seen it on tutorials but we haven't seen it anywhere. 
Can somebody help us ?

Thank you for having read us !

Best regards from Britany !

eglantinepourpre, 2020-11-10:
I'll try the version I've found until anyone can help us ;-)

fa201, 2020-11-17:
Hi,
The PPA is not up-to-date. You may install last release (2.1.3) or build the release candidate (it is not that difficult) with https://librecad.readthedocs.io/en/latest/appx/build.html#building-on-linux.

To switch the Graphical User Interface, you need to select it in Options/Application Preferences. Note that the translation is also better in release candidate.
For more support, please visit the forum.

Good luck.

tormodvolden, 2020-12-17:
There are unofficial PPAs too, and some are up to date:
https://launchpad.net/ubuntu/+ppas?name_filter=librecad

Note from https://help.ubuntu.com/community/PPA

> Subsequently, installing a PPA should be considered to be a low-security alternative as compared to the main repository, but marginally higher security than simply installing software at random from the internet. As part of adding a PPA, you trust the developer to not only install packages, but also to allow them to provide ongoing updates.

lordofbikes, 2021-01-24:
To put it straight, this is about missing PPA maintainer, which is tracked in #894

Concerning the French language, it is supplied with the package. It could be selected on the first start or later in *Options* -> *Application Preferences*

closed 2021-01-24 by lordofbikes