LibreCAD/LibreCAD #1294: Importing .DWG files error

JSpaens, 2020-12-07:
### Expected behavior
Opening .DWG file

### Observed behavior
Receiving warning: "Experimental - Save your work first"
When hitting "Continue" button, nothing happens

### Steps to reproduce or sample file
See attachment

### Operating System and LibreCAD version info
Win10 - 64bit

[Cover.zip](https://github.com/LibreCAD/LibreCAD/files/5651663/Cover.zip)

Version: 2.1.3
Compiler: GNU GCC 4.9.1
Compiled on: Sep 23 2016
Qt Version: 5.4.1
Boost Version: 1.60.0
System: Windows

iomonad, 2020-12-09:
Same problem.

lordofbikes, 2021-01-31:
This file does not crash with latest rc2 code, but the import is empty.

JSpaens, 2021-01-31:
Thanks - I'll check it out. I have AutoCAD installed by now, so the problem
is immaterialized for me...

DuncanLithgow, 2021-02-02:
Here's the file saved as DXF 2013 from Autodesk AutoCAD. The original file is not from an Autodesk licensed application, so it's hard to know the quality of the file.
[Cover_DXF_2013.zip](https://github.com/LibreCAD/LibreCAD/files/5913773/Cover_DXF_2013.zip)

d4mer, 2021-04-01:
Hi All,

I have the same issue on macOS 2.2.0 (not Big Sur though). I try to open a DWG and it fails with the following error:

Cannot open the file
/RC Stuff/airplane blueprints/Plano completo Sinbad 3 metros Export.dwg
Please check its existence and permissions.

The file exists for sure as I was able to open it just fine on 2.1.3.

DuncanLithgow, 2021-04-03:
@d4mer have you tried opening it from different locations? Desktop instead of server for example?

If that's not the problem please open a new issue and if possible attached the .dwg

PrashanthRaju-PS, 2021-04-27:
Hi,

I have the same issue too on Ubuntu (arm64), "Please check its existence and permission". I had created the .dwg with AutoCAD (2021). Attached the screenshots and .dwg file.
[image]
[image]
[AshirvadMainGate.dwg.tar.gz](https://github.com/LibreCAD/LibreCAD/files/6382086/AshirvadMainGate.dwg.tar.gz)

lordofbikes, 2021-04-27:
@PrashanthRaju-PS your file is DWG version AC1032 (ACAD 2018 and newer) and thus related to #1074.
The latest version LibreCAD supports yet is AC1027 (ACAD 2013-2017).