LibreCAD/LibreCAD #1295: Question about memory

JasonStepan, 2020-12-15:
#### Hello admin,
#### When I opened a dxf file with LibreCAD and then closed it, I found that the memory of the LibreCAD process became  larger. I tried this operation hundreds of times and found that the memory keeps increasing. May I ask what is the reason?

tormodvolden, 2020-12-17:
This would typically be due to "memory leaks" in the program, memory chunks are allocated and used, but not released when they are not in use any longer. It could be imperfections in librecad, but could also be in shared libraries used by librecad. Most larger programs have these issues to some extent. What is the amount of growth in your case?

JasonStepan, 2020-12-21:
> This would typically be due to "memory leaks" in the program, memory chunks are allocated and used, but not released when they are not in use any longer. It could be imperfections in librecad, but [...]

About 1M.

JasonStepan, 2020-12-21:
So far, I have found two memory leaks, one is in rs_layerlist.h, there is a private member QList<RS_Layer*> layers, in the RS_LayerList::clear method, the'layers' member is only called the clear method of QList In fact, qDeleteAll(layers.begin(), layers.end()) should be added before clear; the other is in rs_blocklist.h, which has the same memory leak error as rs_layerlist.h.

tormodvolden, 2020-12-21:
Okay, it seems that you know a lot about memory leaks and C++ already :) Maybe you can submit your suggested changes as a pull request?

DuncanLithgow, 2021-01-29:
@lordofbikes are you able to confirm this? Is this something for a 2.2.x roadmap?