LibreCAD/LibreCAD #1300: all text displays as boxes

karlp, 2021-01-07:
### Expected behavior
* opening a dxf with text should display the text
* adding text should make the text visible

### Observed behavior
all text, even dimensions are just outline boxes (see images)
if I drag some text around sometimes it shows the text, but mostly just as outline boxes.
Print preview DOES correctly show the text, and if I edit text, the correct text is there, it's just crazy hard to work like that :)

### Steps to reproduce or sample file
sample file and images attached
[face-place-mods.zip](https://github.com/LibreCAD/LibreCAD/files/5784293/face-place-mods.zip)
[image]
[image]

### Operating System and LibreCAD version info

fedora 32 on x86_64
versions: LibreCAD-2.2.0-rc2 built from source
also librecad-2.2.0-0.5.rc1.fc32.src.rpm the packaged version.

fa201, 2021-01-07:
If draft mode is on (it looks it on) then it is not a bug. Try CTRL+D and check again.

lordofbikes, 2021-01-08:
Indeed, draft mode is on
[image]
https://forum.librecad.org/only-empty-boxes-frames-are-shown-for-image-text-no-line-thickness-td5706737.html

closed 2021-01-08 by lordofbikes

karlp, 2021-01-08:
Might I suggest that draft mode on being the default setting should be revisited?  The out of box experience here is, unfriendly, to say the least.

lordofbikes, 2021-01-08:
I'm not sure if draft mode is on with fresh installation, have to check this.
Anyhow, I link this issue to #1236, maybe we can kill two birds with one stone then.