LibreCAD/LibreCAD #1302: Hatch not displaying.

TheViperShow, 2021-01-10:
[image]
[image]
"Hatch created correctly"
"Hatch area = ..."
But there's clearly no hatch on the plane.
Doesn't display as well when exporting to PDF, it's just like if the hatch was never created.
I tried with a circle, but doesn't work with anything else.

fa201, 2021-01-10:
It works on last build. Please make the **Draft mode** is not on (CTRL+D).
Version: 2.2.0-rc2
Compiler: GNU GCC 7.5.0
Compiled on: Jan 10 2021
Qt Version: 5.9.5
Boost Version: 1.65.1
System: Ubuntu 18.04.5 LTS

If it still does not work, please provide the info shown in **Help/About/Copy** and paste it here.

lordofbikes, 2021-01-10:
I assume the hatch is there, but no line is crossing the circle.
Hatches are build for best results on common paper size.
The hatch area of 0.15 means the circle has ~0.22 radius, really small on common paper size.
So you can try with a bigger circle, e.g. radius = 5.0 .
Or use a smaller scale for the hatch, e.g. 0.05 for this 0.22 radius circle.

TheViperShow, 2021-01-12:
OK thanks.

closed 2021-01-12 by TheViperShow