LibreCAD/LibreCAD #1303: Status bar redimensioning: mouse image is not redimensioned

fa201, 2021-01-10:
### Expected behavior
Mouse image should be redimensioned as the status bar height.

### Observed behavior
When redimensioning the status bar through **Options/Widget options/Status bar**, the mouse image is not redimensioned. 
Size changed from 32 to 30. Test changed from 11 to 8.
[image]

### Operating System and LibreCAD version info
Version: 2.2.0-rc2
Compiler: GNU GCC 7.5.0
Compiled on: Jan 10 2021
Qt Version: 5.9.5
Boost Version: 1.65.1
System: Ubuntu 18.04.5 LTS
Screen definition if relevant : 1920x1080.

Possibly linked to #1010 and #1026.

lordofbikes, 2021-03-14:
This is directly related to #1026 and should be solved with a status bar formatting revamp

lordofbikes, 2022-06-04:
fixed with eb0223968761df405c16a4885fff36c2fd05e9d1

closed 2022-06-04 by lordofbikes