LibreCAD/LibreCAD #1304: Status bar dimensioning: text is not properly vertically aligned

fa201, 2021-01-10:
### Expected behavior
When redimensioning the status bar and the text in the status bar, the vertical centering should be kept.

### Observed behavior
When redimensioning the status bar through Options/Widget options/Status bar, there is more padding (in HTML sense) on the top than on the bottom.
Size changed from 32 to 30. Test changed from 11 to 8.
[image]

But changing the size of status bar to 40 with the same text height does not show the problem !
[image]

### Operating System and LibreCAD version info
Version: 2.2.0-rc2
Compiler: GNU GCC 7.5.0
Compiled on: Jan 10 2021
Qt Version: 5.9.5
Boost Version: 1.65.1
System: Ubuntu 18.04.5 LTS

lordofbikes, 2021-03-14:
This is directly related to #1026 and should be solved with a status bar formatting revamp

melwyncarlo, 2021-11-17:
Also related to #1010, I believe.

Mouse widget problem solved.

lordofbikes, 2022-06-04:
fixed with eb0223968761df405c16a4885fff36c2fd05e9d1

closed 2022-06-04 by lordofbikes