LibreCAD/LibreCAD #1305: Application Crash on macOS Big Sur (11.1)

pawcik, 2021-01-10:
### Expected behavior

After installation application should luinch.

### Observed behavior

First lunch shows the Welcome screen (the one with default Unit, GUI Language and command Language). After pressing OK application is crashing.

### Steps to reproduce or sample file

1. Download version LibreCAD_2-1-3.dmg or LibreCAD-2.2.0-rc2.dmg
2. Install application
3. Skip Welcome Screen 

### Operating System and LibreCAD version info

OS: macOS Big Sur 11.1 (20C69)
LibreCAD: 2-1-3 and 2.2.0-rc2

### Errors
<details>
  <summary>Stacktrace and System Configuration</summary>
  Process:               LibreCAD [76229]
Path:                  /Applications/LibreCAD.app/Contents/MacOS/LibreCAD
Identifier:            com.yourcompany.LibreCAD
Version:               ???
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           LibreCAD [76229]
User ID:               501

Date/Time:             2021-01-10 17:13:16.559 +0100
OS Version:            macOS 11.1 (20C69)
Report Version:        12
Anonymous UUID:        87AA331E-BCD1-62C7-F0B1-2099A9440CEB

Time Awake Since Boot: 190000 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Application Specific Information:
*** Terminating app due to uncaught exception 'NSInternalInconsistencyException', reason: 'Item to be inserted into menu already is in another menu'
terminating with uncaught exception of type NSException
abort() called

Application Specific Backtrace 1:
0   CoreFoundation                      0x00007fff204b36af __exceptionPreprocess + 242
1   libobjc.A.dylib                     0x00007fff201eb3c9 objc_exception_throw + 48
2   CoreFoundation                      0x00007fff204dc512 +[NSException raise:format:arguments:] + 88
3   Foundation                          0x00007fff212986c9 -[NSAssertionHandler handleFailureInMethod:object:file:lineNumber:description:] + 191
4   AppKit                              0x00007fff22c486d3 -[NSMenu insertItem:atIndex:] + 888
5   libqcocoa.dylib                     0x0000000107747d9e qt_plugin_instance + 129998
6   libqcocoa.dylib                     0x00000001077481da qt_plugin_instance + 131082
7   QtWidgets                           0x000000010483c0bd _ZN5QMenu11actionEventEP12QActionEvent + 1021
8   QtWidgets                           0x00000001046f9c68 _ZN7QWidget5eventEP6QEvent + 1224
9   QtWidgets                           0x000000010483a1a1 _ZN5QMenu5eventEP6QEvent + 1105
10  QtWidgets                           0x00000001046b8ec6 _ZN19QApplicationPrivate13notify_helperEP7QObjectP6QEvent + 294
11  QtWidgets                           0x00000001046bb985 _ZN12QApplication6notifyEP7QObjectP6QEvent + 5941
12  QtCore                              0x00000001054b2f24 _ZN16QCoreApplication15notifyInternal2EP7QObjectP6QEvent + 164
13  QtWidgets                           0x00000001046ad568 _ZN14QActionPrivate15sendDataChangedEv + 88
14  QtWidgets                           0x00000001046b24a2 _ZNK12QActionGroup9isVisibleEv + 706
15  QtCore                              0x00000001054e6160 _ZN11QMetaObject8activateEP7QObjectiiPPv + 880
16  LibreCAD                            0x00000001040b3450 _ZN20QC_ApplicationWindow14windowsChangedEb + 64
17  LibreCAD                            0x000000010406d3ca _ZN20QC_ApplicationWindow19slotWindowActivatedEP13QMdiSubWindow + 650
[... 472 lines cut ...]
Foundation                          16K        1 
Kernel Alloc Once                    8K        1 
MALLOC                           231.9M       91 
MALLOC guard page                   32K        7 
MALLOC_NANO (reserved)           384.0M        1         reserved VM address space (unallocated)
STACK GUARD                       56.0M        7 
Stack                             11.0M        7 
VM_ALLOCATE                       1392K       44 
__CTF                               759        1 
__DATA                            13.7M      331 
__DATA_CONST                      11.7M      169 
__DATA_DIRTY                       539K       96 
__FONT_DATA                          4K        1 
__LINKEDIT                       493.8M       31 
__OBJC_RO                         60.5M        1 
__OBJC_RW                         2452K        2 
__TEXT                           216.5M      333 
__UNICODE                          588K        1 
mapped file                      317.8M       69 
shared memory                       40K        4 
===========                     =======  ======= 
TOTAL                              1.8G     1251 
TOTAL, minus reserved VM space     1.4G     1251 

</details>

DuncanLithgow, 2021-03-16:
Could you maybe try and find some other Mac users on the forum and do some testing together?

fa201, 2021-03-16:
Some users reported a success on the forum. Can you please check if this MacOS package solve the issue ? https://forum.librecad.org/2-2-0-rc2-fails-to-install-on-BigSur-11-1-td5719978.html

lordofbikes, 2021-03-16:
That's correct. New packages for BigSur are here and on SourceForge:
https://github.com/LibreCAD/LibreCAD/releases/tag/2.2.0-rc2
https://sourceforge.net/projects/librecad/files/OSX/2.2/preRelease/?upload_just_completed=true
@pawcik when you can confirm that the new package works for you, we can close this issue.

pawcik, 2021-03-18:
Thanks,  I confirm that the latest from @lordofbikes is working for me.

lordofbikes, 2021-03-18:
Many thanks for your reply.

closed 2021-03-18 by lordofbikes