LibreCAD/LibreCAD #1307: Save Print Scale on File

efraletti, 2021-01-16:
After in Print Preview, setting 1:50 scale, fixed and centered to page, saving the file does not seem to store these attributes.

When reopening the drawing, first thing entering Print Preview and the settings are gone.

I observed this in Windows and Unix, and on multiple drawing files.

Could this feature be implemented?

WolfgangAndreas, 2021-01-16:
Hello,
i can confirm this and it would be nice to have this settings saved for future prints.
Regards,
Wolfgang

DuncanLithgow, 2021-01-20:
Please go to Help > About and copy your version information. Please add that to the original post.

Version: 2.2.0-rc2
Compiler: GNU GCC 7.3.0
Compiled on: Dec 31 2020
Qt Version: 5.12.4
Boost Version: 1.65.1
System: Windows 10 (10.0)

I'm seeing slightly different behavior. It is remembering the scale, but showing the paper size incorrectly until i switch scale manually.

efraletti, 2021-01-20:
It is

Version: 2.1.3
Compiler: GNU GCC 10.1.0
Compiled on: Jul  7 2020
Qt Version: 5.15.2
Boost Version: 1.72.0
System: Manjaro Linux

Behavior:

Open LibreCAD > File > Open a .dxf;

Enter Print Preview:

* Print Scale is locked to the current value.
* Scale is 1:56.7568
* Fitted to Page

Then I have to
* Unfix Print Scale
* Enter 1:50 Scale
* Center to Page
And the job is ready to print.

I have tried saving the file now, with or without fixing Print Scale locked to the current value, in this system and in Windows with the latest release of LibreCAD and got the same results.

It seems the attribute is not carried on the .dwg file metadata?

Thanks.

DuncanLithgow, 2021-01-21:
I expect you're right that it is not carried in the DXF/DWG file. Both use paper space to manage what you're struggling with. Is this known and expected behaviour @lordofbikes that LibreCAD doesn't remember print preview setting for a file between sessions?

gerardbm, 2021-01-30:
This issue was fixed. Please, read this threads:

https://github.com/LibreCAD/LibreCAD/issues/705
https://github.com/LibreCAD/LibreCAD/pull/1050

efraletti, 2021-01-31:
thank you, i installed librecad-git and it behaves like expected now.

closed 2021-01-31 by efraletti