LibreCAD/LibreCAD #1308: How To Install After Make From Source on Linux

JavaScriptDude, 2021-01-17:
I have just finished installing LibreCad from Master on Github on a Ubuntu 20.04 system. The make went well and I could find the desktop files and executable and libraries under unix directory. However, I could not find any instructions on how to install the desktop file and get all the files to be properly deployed.

I had to do it all by hand and it caused some issues right away with font detection that I had to figure out. I'm sure there are some more hand install gotchas I have not hit yet.

Are there scripts or instructions on how to properly install after a make on Linux from Source?

fa201, 2021-01-17:
Yes there are some : https://librecad.readthedocs.io/en/latest/appx/build.html
As far as I know there is no desktop file: LibreCAD has to be launched through console.

JavaScriptDude, 2021-01-18:
Nice. Actually there is a desktop file now under `./desktop`  folder.

Although, there is no icon in png format here are steps to deploying on linux (WIP):
```
# Make desktop icon (Depends on imagemagic)
convert "./desktop/graphics_icons_and_splash/Icon LibreCAD/LibreCAD_Icon 256x256.ico" $HOME/.local/share/icons/librecad.png

# Make Home for App
mkdir -p ~/apps/

# Copy app over to final home
cp -r ./unix ~/apps/librecad

# Make symlink to executable
sudo ln -s $HOME/apps/librecad/librecad /usr/local/bin/librecad

# Deploy Desktop file
desktop-file-install --dir=$HOME/.local/share/applications ./desktop/librecad.desktop
```

The one gotcha so far is after opening LibreCad for the first time, you need to go to `Options -> Application Preferences -> Paths` and set `Fonts` to `<home_path>/apps/librecad/resources/fonts` . Without doing this, there will be no text displayed in drawings and the font drop down will be empty throughout the app. I assume that this can be fixed automatically in the bootstrap of LibreCad to automatically resolve this path and set it if its blank.

closed 2021-01-18 by JavaScriptDude