LibreCAD/LibreCAD #1310: [FEATURE REQUEST] Import from .pdf

grappas, 2021-01-23:
Simple as it is.

melwyncarlo, 2021-11-16:
Overly complicated. You'd need to either find a relevant library, or implement it from scratch. 
Question is: do you parse the PDF for vector graphics, or do you parse the entire PDF as image?
It's much more feasible to convert PDF to image format like SVG using online tool, library, Python, or Inkscape, 
and then import the image file into LibreCAD.

u2fly, 2025-04-14:
@dxli @sand1024 

- https://forum.librecad.org/POLYLINE-PDF-2-Bugs-in-Api-tp5726819p5726832.html
> add a PDF? Hmm... how and where? Could you please expand on this?
>
> UPD: Oh, noticed this in your video. I suppose we should apply more strict to image format. PDF is definitely is not something correct that is expected there, AFIK pdf is not supported at all ... 

Few options:
- https://wiki.qt.io/Handling_PDF

__Qt6 Qt PDF__ module could render PDF page to QImage buffer
- https://doc.qt.io/qt-6/qtpdf-index.html
> If you only need to render page images, without features such as text selection, search and navigation, this module includes a [QImageIOHandler](https://doc.qt.io/qt-6/qimageiohandler.html) plugin that treats PDF as a scalable [image format](https://doc.qt.io/qt-6/license-changes.html#qt-6-1-2-qt-image-formats-172), similar to [SVG](https://doc.qt.io/qt-6/qtsvg-index.html). You can simply use [Image](https://doc.qt.io/qt-6/stylesheet-reference.html#image), and set the [currentFrame](https://doc.qt.io/qt-6/qml-qtquick-image.html#currentFrame-prop) property to the page index that you wish to display. If the PDF file does not render its own background, the image has a transparent background.
- https://doc.qt.io/qt-6/qpdfdocument.html#render
>Renders the page into a [QImage](https://doc.qt.io/qt-6/qimage.html) of size imageSize according to the provided renderOptions.

__Poppler Qt6__ module also could do that
- https://poppler.freedesktop.org/api/qt6/
> Pages can be rendered to QImages