LibreCAD/LibreCAD #1313: Working on a drawing - - - - cannot see anything displayed in the layer list

dabeegmon, 2021-01-28:
### Expected behavior
When I add a layer I would  expect to see something in the layer list

### Observed behavior

Layer list box is blank

### Steps to reproduce or sample file

I tried to grab a screenshot but I need to trim it to give only what I want to send to you. 
(I've got 4 - 1920x1080 monitors so there is a lot of irrelevant information on the screenshot!)
The drawing is on 2 of the monitors.

### Operating System and LibreCAD version info

Debian testing (11) on LXQT and LibreCad version 2.1.3

lordofbikes, 2021-01-28:
There should be at least layer **0** which is always there and could not be deleted.
So it is not clear what's going on.
Make a screenshot of the LibreCAD window only or use e.g. Gimp to crop at least the layer list widget.
Otherwise it's hard to give good advice.

dabeegmon, 2021-01-28:
On Thu, Jan 28, 2021 at 7:22 AM Armin Stebich <notifications@github.com>
wrote:

> There should be at least layer *0* which is always there and could not be
> deleted.
> So it is not clear what's going on.
> Make a screenshot of the LibreCAD window only or use e.g. Gimp to crop at
> least the layer list widget.
>
tried gimp and that was working like a concrete canoe!
I'll try a bit more - - - - what detail will show the most?
The layer box, the drawing  ?????

DuncanLithgow, 2021-01-29:
@dabeegmon are you familiar with what the Layer list should look like? Here is the layer list of a fresh file 
[image]

It shows one layew with these attributes: Visible, Not Locked, Printable, Not a construction Layer, Color is black and the layer is named "0". Is that nothing like what you are seeing?

dabeegmon, 2021-01-30:
On Fri, Jan 29, 2021 at 4:10 PM Duncan Lithgow <notifications@github.com>
wrote:

> @dabeegmon <https://github.com/dabeegmon> are you familiar with what the
> Layer list should look like? Here is the layer list of a fresh file
> [image: billede]
> <https://user-images.githubusercontent.com/3223028/106332154-84516a80-6286-11eb-9532-f17c1ec2f083.png>
>
> It shows one layew with these attributes: Visible, Not Locked, Printable,
> Not a construction Layer, Color is black and the layer is named "0". Is
> that nothing like what you are seeing?
>
>
> Serious weirdness - - - - when last working I could see nothing  except as
above on the layer list.
See attached for the way it is looking now (did a shutdown most programs
and then a reboot and just now restarted LibreCAD).
That is what I was expecting - - - - except all I saw was as above - - - -
- dunno as to why!

Suggestions which one of the /var/log files to poke into for clues?

Thanking all for their consideration and assistance!!!

(Anyone know how to not only make a rectangle but replicate it ever few
hundred millimeters? floor joists - - - - only will have 24 of them- - -
argh!)

Regards

lordofbikes, 2021-01-30:
Which attachment? You can simply drag image files to the reply box here to post it.
There are no `/var/log` files that can help here, show us what you see.

Make the rectangle a block, then you can repeat it in a matrix by adjustable X/Y count and distance.
You will find how to do this in the forum https://forum.librecad.org using the search function there.
Search for block or matrix.
There is also https://librecad.readthedocs.io/ with search function too.

dabeegmon, 2021-01-30:
On Sat, Jan 30, 2021 at 2:15 AM Armin Stebich <notifications@github.com>
wrote:

> Which attachment? You can simply drag image files to the reply box here to
> post it.
> There are no /var/log files that can help here, show us what you see.
>

Sorry - - - - that's unusual behavior for me.

This is what I see now. Saw only the blank block previously.

[image: screenshot.png]

lordofbikes, 2021-01-30:
Feel free to open this issue again when you have posted an image which we can see.

closed 2021-01-30 by lordofbikes

natrius, 2021-04-19:
I'm commenting because: Is it possible that you have selected the image and cannot see the lines anymore, @dabeegmon ? If i import a image and click somewhere beside my lines, the image is selected and the lines are getting hidden. A second left mouseclick would reveal the lines again. 

(Was looking for an issue about the "select image when clicking" thing, but creating a new layer for the image, switching the layer of the image to the new created layer and locking that layer did the thing for me)

lordofbikes, 2021-04-19:
This issue is about an empty layer list, not images hiding other entities.
But the reporter was not able to post a screenshot for further investigation.

cpolindara, 2025-03-18:
> [@dabeegmon](https://github.com/dabeegmon) are you familiar with what the Layer list should look like? Here is the layer list of a fresh file [image]
>
> It shows one layew with these attributes: [...]

I experienced something similar when opening some dxf files and figured out that LibreCAD accepts two types of dxf files, Drawing Exchange and QCad 1.x files, both with the .dxf extension. I had been selecting the second one and hence nothing was displayed. When i selected the first type of file everything worked. I don't know if it is related but here i leave my two cents.