LibreCAD/LibreCAD #1316: Application Crash when opening the dwg

Ubuntu-2020, 2021-01-31:
When you open a [**dwg** file](http://www.dwgindir.com/puansiz/puansiz-2-boyutlu-dwgler/puansiz-yapi-ve-binalar/asma-katli-cafe-projesi.html), **LibreCAD 2.0.9** closes automatically!
You can also download the **dwg** file from **[here](https://wdfiles.ru/8465ab06c9f5c660)**.

DuncanLithgow, 2021-01-31:
I can't download that file from the links you've given.

I suggest you install a more recent version. There have been problems with DWG import in earlier versions. v2.2rc2 is the best currently for stability of DWG imports. Please go to the forum if you need help with that.

Please test the file again and let us know what happens.

Source files: https://github.com/LibreCAD/LibreCAD/releases
Install from source: https://wiki.librecad.org/index.php?title=Build_from_source#Building_LibreCAD_2.0_on_Linux
Forum: https://forum.librecad.org/

Ubuntu-2020, 2021-01-31:
>@DuncanLithgow I can't download that file from the links you've given.

[File sharing service](https://wdfiles.ru/8465ab06c9f5c660):
[image]

lordofbikes, 2021-01-31:
I've downloaded the file.
This issue is already tracked in #988

closed 2021-01-31 by lordofbikes