LibreCAD/LibreCAD #1317: LibreCAD does not see the drawing if its extension is DWG

Ubuntu-2020, 2021-01-31:
**LibreCAD 2.0.9**.
The folder contains the drawings. Most drawings have an extension of **dwg**. But some drawings have the extension **DWG** that is written in capital letters.
For example:
- window.dwg
- chair.DWG

Run **LibreCAD 2.0.9** and open the folder with the files and see only the file there **window.dwg**
Thus **LibreCAD 2.0.9** does not see the file **chair.DWG**

DuncanLithgow, 2021-01-31:
Please test with a more recent version of LibreCAD and reopen this bug if the problem still exists.

closed 2021-01-31 by DuncanLithgow

Ubuntu-2020, 2021-01-31:
@DuncanLithgow @lordofbikes
In **Zorin OS** - **LibreCAD 2.1.2** the same situation!

lordofbikes, 2021-01-31:
LibreCAD 2.0.9 has no DWG support, so there is no need to list DWG files.
LibreCAD 2.1.2 should support DWG files, but you must change the file type box in the open dialog from DXF to DWG to list DWG files.

Ubuntu-2020, 2021-01-31:
@lordofbikes
It's about writing the file! Not in its format. If the last letters of the extension are written in capital letters, then LibreCAD will not see such drawings!
Example: create an empty folder and copy a single drawing in **dwg** format to it. Then copy this drawing. Then rename the last letters of the extension by writing them in capital letters **DWG**.
This way you will have only two drawings in the folder, both in dwg format. Only one drawing will end with **lowercase letters dwg**, and the other with **uppercase letters DWG**.
Example:
- window.dwg
- window-copy.DWG

Then open this folder in LibreCAD 2.1.2 and you will see that the program sees only **one drawing**, in which the last letters of the **dwg** extension are written in **lowercase letters**!

lordofbikes, 2021-01-31:
follow up #791