LibreCAD/LibreCAD #1319: converted lff font files do not work.

queenmedley, 2021-02-02:
The ttf(DejavuSans) file was converted to lff font using ttf2lff with archlinux
It looks well made, no error or warning message.

 > [wonhae@office1:~]$ ttf2lff DejaVuSans.ttf DejaVuSans.lff
> TTF file: DejaVuSans.ttf
> LFF file: DejaVuSans.lff
> Family:    DejaVu Sans
> Height:    2384
> Ascender:  1901
> Descender: -483
> Factor:    0.00602813
> 

When entering text in the librecad program, I can see the font name created in the font list.
However, even if I select that font, but it was the same as ISO.
I posted this issue on http://forum.librecad.org but it seems to be an issue with ttf2lff.**

### Expected behavior
Font style should be changed 

### Observed behavior
Font style not changed, just same as iso 

### Steps to reproduce or sample file
ttf link
https://drive.google.com/file/d/1bBgAQX7jTXWVQMn2aFbFbK9C3WcXUOXA/view?usp=sharing
lff link
https://drive.google.com/file/d/1HcFDLVtTKHA7Kw3XY-oqa0dHprTTTL0e/view?usp=sharing

### Operating System and LibreCAD version info
Os : Archlinux, 
librecad ver : 2.13, compiler Gcc 10.1.0, qt 5.15.2

lordofbikes, 2021-02-07:
ttf2lff uses [FreeType](https://www.freetype.org/index.html) to convert each character of a TTF font to a LFF font glyph.
What you have got is a fallback implementation when something went wrong.
I can't say yet, if this is an issue with ttf2lffs interface implementation to FreeType or if it is an inability of FreeType to interpret DejaVuSans.ttf.
The implementation of ttf2lff is very old, maybe it needs to be reworked.
Anyhow this needs some deeper investigations in [tools/ttf2lff/main.cpp](https://github.com/LibreCAD/LibreCAD/blob/master/tools/ttf2lff/main.cpp) and FreeType.

lordofbikes, 2021-04-04:
This finally turned out as LibreCAD bug.
To solve for any older versions, simply rename the LFF font file to lower case, then it should work.
Fixed with https://github.com/LibreCAD/LibreCAD/commit/56dddeb332009c7772ed100e66a89f41713260d2

closed 2021-04-04 by lordofbikes