LibreCAD/LibreCAD #1322: Suggestion: Adding info on selected objects to statusbar

matthijskooijman, 2021-02-24:
I recently got started with LibreCAD to edit an existing drawing, which was exported from vectorworks and quite complex (with a lot of nested blocks). That complexity, combined with my inexperience with LibreCAD, made it hard to figure out how this drawing was structured.

One thing that I would think could help here, is to add some info to the status bar about the currently selected object(s), similar to what Inkscape does. The most useful info would be the number and types of objects selected, plus their layer. When just a single insert is selected, I would also suggest showing additional info, such as the number of points in a polyline, or the name of the block that is inserted.

For example, you would get something like:

> selected 3 objects of type Insert in layer foo
> selected 5 objects of type Insert, Line, Circle in layer 0
> selected 5 objects of type Insert, Line, Circle in 2 layers
> selected Circle in layer bar
> selected Insert (Name of block here) in layer bar
> selected Polyline of 8 points in layer 0

One open question is where to put this info exactly. A third line on the statusbar would be easiest, but also takes up additional space. Maybe that extra line could be separately toggled?

lordofbikes, 2021-02-24:
Number of selected entities and their over all length is always available in the bottom status bar.
The other information is available with the **List** plugin.

matthijskooijman, 2021-02-24:
I tried the list plugin, but for me it just stays stuck on "Select objects" with a cross pointer, nothing happens when I click or drag around objects, but that's another issue I guess. Still, it seems that the UX for that plugin is to start the plugin and then select things, while IMHO it would be a lot more convenient to 1) Have something that's always shown while you do work, and 2) applies to the current selection, not a selection to be made specifically for the plugin (but maybe the plugin can do that as well if it is used as intended, dunno).

lordofbikes, 2021-02-24:
This is common behavior in LibreCAD.
When nothing is selected, you are asked to select entities and finish your selection with **ENTER**. 
Then the plugin window will appear.
If there are already selected entities, the plugin opens immediately with information about the selection.

Another possibility is the **Properties** tool from **Modify** menu/tool bar.
It opens a window for one clicked entity, if it is selected or not.

What you describe sounds like a dockable property editor.
This is something we will have in LibreCAD 3 one day, but I assume that this will not find its way to LibreCAD 2.

matthijskooijman, 2021-02-24:
Ah, thanks, I didn't know about the enter, that helps. Also, because of #1323, my clicks were made doubleclicks, so I ended up selecting and immediately deselecting objects, making it look like nothing happened at all. Using a mouse, and knowing about the enter, the list plugin indeed seems to work as intended, thanks!

The properties tool and the list plugin together allow getting the information I'm looking for (especially the list plugin now lets me figure out the block that is inserted by an insert, which I didn't know how to get at yet), but I still feel that having this info more readily available in the statusbar would be helpfull. Also, my suggestion would summarize info for a selection of multiple objects, rather than listing all individually (like the list plugin) or limiting to just one object (like the properties tool).

I realize that whether or not this is useful is very much a matter of taste and workflow, so something dockable or otherwise optional would indeed make sense (though to accomodate new users, I would suggest having it enabled by default).

> What you describe sounds like a dockable property editor.

Well, it's not so much an editor, but a condensed and (because of that necessarily) read-only summary, but dockable sounds good :-)

>  This is something we will have in LibreCAD 3 one day, but I assume that this will not find its way to LibreCAD 2.

Yeah, I was halfway through writing this issue when I found out about LibreCAD 3, but I was unsure about its status, so it seemed better to just share my thoughts here first.

thecadman2017, 2021-02-24:
Hi I am Manny Monroy Vietnam Vet and wondering if there is any discounts
for veterans to get the cad program. I was cut off from Autodesk Inventor
and am very bored without it.
Can you help me out.
Thank you
Manny Monroy
Don't Ever Give Up !!!

fa201, 2021-02-24:
@thecadman2017, LibreCAD is free.

DuncanLithgow, 2021-03-16:
@thecadman2017 librecad doesn't have much similarity to Inventor. If you're looking for 3D, then you'd best look to FreeCAD.

DuncanLithgow, 2021-03-16:
Sounds to me like this is not a bug and not a planned feature as it is already implemented in various ways. Reopen if I'm wrong.

matthijskooijman, 2021-03-16:
@DuncanLithgow, I can imagine that an addition to the statusbar as I suggested would still be useful in LibreCAD 2, even if a full dockable status area/editor or whatever more fancy version of this request is subject of changes in LibreCAD 3. However, I can't comment on whether this is something others would also consider useful and/or would care to implement. When I wrote this issue, I had in mind to maybe provide a patch for this, but now I suspect that 2.x development is not really moving much in favor of 3.x, but 3.x is not yet usable for my usecases, so I'll likely spend my limited time elsewhere, and leave it up to others to decide whether this should remain open or be closed :-)

sand1024, 2026-09-26:
supported vie properties , status bar etc

closed 2026-09-26 by sand1024