LibreCAD/LibreCAD #1323: Inserting block from library browser inserts it twice

matthijskooijman, 2021-02-24:
### Expected behavior
When I insert a block from the library browser, I expect to get one block in the block list for the drawing, and one entitity in my drawing.

### Observed behavior
I get two blocks in the block list, and two entities (one for each block). This duplication is also present in the saved DXF file.

### Steps to reproduce or sample file
 - Create a new file
 - Select any block in the library browser
 - Click "insert"
 - Click in the drawing to place the block
 - Notice you get two blocks in the block list: 
[image]
 - Select the inserted block in your drawing and move it, notice that there is a second one underneath: 
[image]

### Operating System and LibreCAD version info
I tested with today's git master, but the same problem already exists in 2.1.3 from Ubuntu.

Version: 2.2.0-rc2-1-g6dfb387c
Compiler: GNU GCC 9.3.0
Compiled on: Feb 24 2021
Qt Version: 5.12.8
Boost Version: 1.71.0
System: Ubuntu 20.04.2 LTS

lordofbikes, 2021-02-24:
This is the implemented behavior, in case you want to apply different properties (angle, scale) to the inserts.
When you want multiple copies of **blk1**, then first insert one instance from the library browser.
Then add multiple copies of **blk1** with insert from the drawing block list.
There is no block duplication then and you can modify all instances at once.

matthijskooijman, 2021-02-24:
Ah, but the problem also occurs when I click just once.

Closer inspection shows that this only happens when I use my Wacom tablet pen. When I use the touch (i.e. a regular mouse pointer), it inserts it just one.

So I think this actually completely unrelated to the blocks, but a more general bug where my tablet pen registers clicks twice. That also explains why trying to place a circle with the two-point method shows an error message at the first click, since that probably just registers as two clicks. Hm.

matthijskooijman, 2021-02-24:
I added some debug prints to the code and confirmed that a click with my pen registers as a mouse click and a tablet event. In particular, both of these fire for a single click:

https://github.com/LibreCAD/LibreCAD/blob/6dfb387cdf927d69ec75a0dc445b574f860bd9b3/librecad/src/ui/qg_graphicview.cpp#L257
https://github.com/LibreCAD/LibreCAD/blob/6dfb387cdf927d69ec75a0dc445b574f860bd9b3/librecad/src/ui/qg_graphicview.cpp#L408-L411

And since the tablet event calls the mouse event, that one gets called twice:

```
tabletEvent press 2 92
mouseEvent press 1
mouseEvent press 1
tabletEvent release 2 93
mouseEvent release 1
mouseEvent release 1
```

```diff
@@ -256,6 +257,7 @@ void QG_GraphicView::resizeEvent(QResizeEvent* /*e*/) {
 
 void QG_GraphicView::mousePressEvent(QMouseEvent* event)
 {
+    printf("mouseEvent press %u\n", event->button());
     // pan zoom with middle mouse button
     if (event->button()==Qt::MiddleButton)
     {
@@ -288,6 +290,7 @@ void QG_GraphicView::mouseDoubleClickEvent(QMouseEvent* e)
 void QG_GraphicView::mouseReleaseEvent(QMouseEvent* event)
 {
     RS_DEBUG->print("QG_GraphicView::mouseReleaseEvent");
+    printf("mouseEvent release %u\n", event->button());
 
     event->accept();
 
@@ -406,10 +409,12 @@ void QG_GraphicView::tabletEvent(QTabletEvent* e) {
         case QTabletEvent::Stylus:
         case QTabletEvent::Puck:
             if (e->type()==QEvent::TabletPress) {
+                printf("tabletEvent press %u %u\n", e->device(), e->type());
                 QMouseEvent ev(QEvent::MouseButtonPress, e->pos(),
                                Qt::LeftButton, Qt::LeftButton, Qt::NoModifier);//RLZ
                 mousePressEvent(&ev);
             } else if (e->type()==QEvent::TabletRelease) {
+                printf("tabletEvent release %u %u\n", e->device(), e->type());
                 QMouseEvent ev(QEvent::MouseButtonRelease, e->pos(),
                                Qt::LeftButton, Qt::LeftButton, Qt::NoModifier);//RLZ
                 mouseReleaseEvent(&ev);
```

It seems this behavior is intentional, looking at [the docs](https://doc.qt.io/qt-5/qtabletevent.html):

> The event handler QWidget::tabletEvent() receives TabletPress, TabletRelease and TabletMove events. Qt will first send a tablet event, then if it is not accepted by any widget, it will send a mouse event. This allows users of applications that are not designed for tablets to use a tablet like a mouse. However high-resolution drawing applications should handle the tablet events, because they can occur at a higher frequency, which is a benefit for smooth and accurate drawing. If the tablet events are rejected, the synthetic mouse events may be compressed for efficiency.

And `QG_GraphicView::tabletEvent()` processes, but does not accept the event. A quick test adding `e->accept()` shows that that indeed prevents the subsequent mouse events from being sent, fixing this problem.

I'm not sure if this is the best place for it, but this helps for me:

```diff
@@ -382,6 +385,7 @@ bool QG_GraphicView::event(QEvent *event)
  */
 void QG_GraphicView::tabletEvent(QTabletEvent* e) {
     if (testAttribute(Qt::WA_UnderMouse)) {
+        e->accept();
         switch (e->device()) {
         case QTabletEvent::Eraser:
             if (e->type()==QEvent::TabletRelease) {
```

app4soft, 2024-08-27:
In latest LibreCAD `2.2.2_alpha` <kbd>Middle Mouse Button</kbd> click acts both as "Zoom Paning" and "Zoom Out" simultaniously.

Not sure is it related directly to this issue, or should be created a separate issue.