LibreCAD/LibreCAD #1324: Suggestion: Improved block list navigation

matthijskooijman, 2021-02-24:
I recently got started with LibreCAD to edit an existing drawing, which was exported from vectorworks and quite complex (with a lot of nested blocks). That complexity, combined with my inexperience with LibreCAD, made it hard to figure out how this drawing was structured.

One thing that would be helpful, is if the block list was more adaptable, allowing to show a subset of blocks, or show a hierarchical structure of blocks to help understand what blocks are used where and how.

In particular, I would think of two additional options for the block list. The first, to limit the blocks that are shown. This could be implemented as a few checkboxes, each toggling a category of blocks to be shown in the list (note that I'm not talking about changing the visibility/freeze status of these blocks in the drawing, just whether the block is displayed in the list). Categories that I would think are useful:
 - Blocks directly inserted into the current drawing (or block when editing a block)
 - Blocks indirectly inserted (i.e. inserted into blocks that are inserted into the current drawing/block, recursively)
 - Unused blocks

The default setting (which matches the current behavior) would be to have all categories enabled.

A second option would be to add (optional) hierarchy to the list. I can imagine two modes:
 - Flat list (matching current behavior)
 - Hierarchy based on blocks inserted into other blocks. This shows the same blocks as the flat list (possibly filtered by the option described above), but below each block, you would list any blocks that are inserted into that block. This would happen recursively, so any blocks inserted into those blocks are listed below them, etc. This means that blocks will probably be shown in the block list multiple times.

Three other small improvements related to blocks that I thought of are:
 - Allow selecting all instances of a block in the current drawing
 - Allow selecting/finding a block in the block list based on a (single) selected insert in the drawing
 - Add the block name to the properties window for an insert object, maybe even as a dropdown so you can change the block that is being inserted by that insert.

matthijskooijman, 2021-02-24:
Note, showing only unused blocks probably makes it easy to prune all unused blocks, as requested in #951.

dellus, 2021-02-24:
To prune all unused blocks see here   https://forum.librecad.org/Remove-the-empty-layers-tp5716976.html

leo-smi, 2021-03-05:
Also, when we have a lot of blocks, the LibreCAD freezes sometimes, the CPU load go to 100% and is very slow to open the file.

u2fly, 2025-04-16:
@dxli 

Block list "Filter" not shows in filter results blocks with `[ ] ( ) < >`/etc. symbols in the name (making "Filter" fully unusable in LFF font editing, as all blocks has `[ ]` brackets)

u2fly, 2025-04-21:
@sand1024 

> Block list "Filter" not shows in filter results blocks with [ ] ( ) < >/etc. symbols in the name (making "Filter" fully unusable in LFF font editing, as all blocks has [ ] brackets)

Also, adding to this LFF blocks filtering issue: filtering shows empty block list, and after clean up filter input field there are no blocks listed at all.

And suggestion on new LFF file reading with blocks arranged in an array/table: 
- set it to 16-blocks-wide (now 10) to fit hexadecimal Unicode table width

sand1024, 2025-04-21:
@u2fly 
Well, block widget as well as library widget needs refreshing indeed and I'll plan to review them. 

Well, filtering is historical thing, and, in more general sense - as for LFF - I fill that some separate more of Block Widget is needed to support fonts editing.  Especially if the amount of chars in the font is significant. 

@set it to 16-blocks-wide (now 10) to fit hexadecimal Unicode table width

The amount of columns is controlled by the setting here: Application Preferences -> Render ->Font Files....

Or probably I've missed your point?

u2fly, 2025-04-21:
> The amount of columns is controlled by the setting here: Application Preferences -> Render ->Font Files....

Ok, I see it now.

> Or probably I've missed your point?

I suggest set it to 16 by default.