LibreCAD/LibreCAD #1326: Change default linewidth to 0.25 mm

dellus, 2021-03-06:
The default linewidth for layer 0 and all newly created layers is 0.00 mm.
New inexperienced users often are drawing with this setting without being conscious of it. On screen everything looks all right but when printing on paper or exporting to pdf often entities with linewidth 0.00 mm don't appear at all or are very faint, leaving despaired users. The behaviour varies with different LibreCAD versions, OS's, pdf-viewers, and printers or their drivers .
I suggest to change the default linewidth for layer 0 and all newly created layers to 0.25 mm. This is a widely accepted standard for a normal line. Experienced users who might need 0.00 mm (for CAM?) will know what to do.  
By the way, there is also an option to set "default" linewidth. It turns out to be the same as 0.00 mm. But there is no possibility in LibreCAD to define this default setting. In other CAD packages you can set this standard, and it only makes sense then.

dellus, 2021-11-24:
Thanks, melwyncarlo. This will spare us a lot of grief.

melwyncarlo, 2021-11-24:
You're welcome. :smiley:

jbergengh, 2021-11-24:
How will this work with layers?  If I create a layer with a default line width of (something), and create a new line on that layer, will the line width be .25mm or get the layer's default line width?

jbergengh, 2021-12-10:
@melwyncarlo can you double check to verify this fix is working? I downloaded the source zip for Task_22 and compiled it.  Newly created layers still show a line width of 0mm, as does layer zero.

GSStnb, 2021-12-10:
A little late to this, but a defined line width for layer 0 might cause a problem with blocks.  Layer 0 is a special case as it it used for blocks where line attributes should be undefined, or 'By Layer' (unless attributes are required for a specific purpose).  Blocks (drawn on Layer 0, or any other layer) will use assigned attributes when imported into a drawing.  If left undefined, a block will adopt the attributes of the layer it is being imported into (preferred behaviour).

Generally speaking, Layer 0 should not be used for general drawing purposes.

jbergengh, 2021-12-10:
@GSStnb good point.  Layer 0 is a special case for blocks and its attributes should remain undefined.

I think @dellus has a very good point though.  Inexperienced users will likely start drawing on the default layer 0, then run into the problems later (printing, PDFs, etc) as in the OP.

How about this:  Whenever a new drawing is created, what if LibreCAD created a new default, active layer, maybe called "unnamed layer", in addition to Layer 0.  The "unnamed layer", and any additional layers created by the user, would default to a line width of .25mm.  This would solve both problems.  The inexperienced user would be drawing on a layer with a line width of .25mm, and Layer 0 would remain as-is with no attributes for use with blocks.

GSStnb, 2021-12-10:
A drawing layer separate from Layer 0 would reasonable, but perhaps it doesn't need to be a programmatic change (*).  When LC starts, it reads the default template (../library/templates/empty.dxf ?).  That template can be defined with an additional drawing layer that includes the prescribed line attributes.  User defined drawing templates (Application Preferences -> Paths -> Template) can include additional layers / attributes, and that template should (does) take precedence over the default temple). 

*: Is the default template included with the installation software package, or is it created the first time LC is run?

dellus, 2021-12-10:
I have looked how it is done in other AutoCAD like apps.
In NanoCAD (a russian clone) out of the box the layer pen attribute is "Default". The value of "Default" is defined elsewhere in general app settings, out of the box it is 0.25mm. 
In FreeDraft, an other clone, rather simple, it's all the same.
I don't have access to AutoCAD, nor do I have any experience, but I have found this at knowledge.autodesk.com for AutoCAD 2020:
"Lineweight values consist of standard settings including BYLAYER, BYBLOCK, and DEFAULT. The DEFAULT value is set by the LWDEFAULT system variable, which has an initial value of 0.01 inches or 0.25 mm. All new layers use the default setting."  
LibreCAD does have the line width setting "Default",  but no facility to define it, it turns out to be 0.00mm in effect. 
I wonder if there is any difference between using layer pen attributes linewidth "Default" (at 0.25mm) and directly "0.25mm".

dellus, 2021-12-10:
I have to add: In NanoCAD and FreeDraft the described also applies to Layer 0.

dellus, 2021-12-10:
"I wonder if there is any difference between using layer pen attributes linewidth "Default" (at 0.25mm) and directly "0.25mm"."
I think I have understood now. "Default" is defined in what is "Application Preferences" in LibreCAD, not in the drawing file.

dellus, 2021-12-10:
Sorry for another post, but it comes to my mind one after the other.
Could it be @GSStnb has got something wrong?  " Layer 0 is a special case as it it used for blocks where line attributes should be undefined, or 'By Layer' (unless attributes are required for a specific purpose). Blocks (drawn on Layer 0, or any other layer) will use assigned attributes when imported into a drawing. If left undefined, a block will adopt the attributes of the layer it is being imported into (preferred behaviour)."
You can not set Layer pen attributes to 'Undefined' or 'By Layer'.  Or have I got something wrong?

GSStnb, 2021-12-10:
You are correct @dellus.  It is the _entities_ attributes that are "By Layer", "By Block", or "Unchanged" - that is my mistake.  

The important part is that blocks will adopt the current layer's attributes when inserted, unless the block's attributes are set to something specific when they are created, such as a line-width of 0.25 mm, colour, etc...

The gist is that the if the line width, or any other attribute, is assigned to Layer 0, the entities comprising the block will adopt those values and it is those values that will be used when inserting a block in a drawing.  Defining specific attributes for a block can create problems, e.g. when a block is inserted in a drawing with a wildly different scale - the line can end up much too thick (I did it in one of my own drawings).  Sometimes setting specific attributes for a block would be necessary, but not often (?).

Ref: https://forums.autodesk.com/t5/autocad-forum/what-is-layer-0-for/td-p/57852

 Having said all that, I should actually try setting the line thickness in layer 0, and create a block, etc...

fa201, 2021-12-10:
I almost never use blocks but I am a bit afraid by the outcomes of the discussion with blocks.

I think we should also think about drawings made with older version of LibreCAD: if hard-coding a change of line width on layer 0 could mess *old* drawing then it could be a really bad surprise for those who upgrade LibreCAD.
It looks that changing the template be safer since every one can change its parameter whereas almost nobody can change the code...

Another solution could be a check before printing and / or saving: a *warning* window would inform the user that a layer (apart from 0) has 0mm line-width and that it makes the line non visible for printing. Then LibreCAD remains flexible for experienced user and provide an additional guard for inexperienced user. This is also backward compatible...

dellus, 2021-12-10:
@fa201:
As I imagine it, drawings made with older version of LibreCAD will open in a new version with the layers retaining the saved line width, as this is written into the drawing file.

jbergengh, 2021-12-11:
@fa201 I totally agree.  Let's not mess with layer 0 at all.

I just took a look at the default template (/resources/library/templates/empty.dxf).  It's got five layers in addition to layer 0 in it.  **All five of those layers are set to pen width = 0!**  Clearly that is setting the inexperienced user up for disaster.  Nothing's going to print from any of those layers by default.  All five layers in empty.dxf should be set to .25 mm.

I think we also need two coding changes.

1. If the template "empty.dxf" does not exist, and no other user template is defined in the preferences, LC creates a new drawing with just layer 0 in it.  It should also create a "noname" default layer with pen width = .25 mm.  This will steer the user towards not drawing on layer 0 and get the inexperienced user a better default pen width.

2. When a user creates a new layer, the default pen width in the Layer Settings dialog box should be .25 mm instead of 0, which the user can change via the drop down menu before he clicks OK.

melwyncarlo, 2021-12-18:
> @melwyncarlo can you double check to verify this fix is working? I downloaded the source zip for Task_22 and compiled it. Newly created layers still show a line width of 0mm, as does layer zero.

@jbergengh I'm sorry, I just realised that I had forgotten about amending (just a tiny word change) another file. The change basically ensures that the layer width of layer 0 is `default`, and not `0.0 mm`. It should work all fine now. I have just tested `Task_22`.

dxli, 2026-08-19:
already fixed in master(2.2.2)

closed 2026-08-19 by dxli