LibreCAD/LibreCAD #1327: Save a single layer as a dxf

CaseyB, 2021-03-12:
I want to be able to create a drawing with multiple layers and export them into individual dxf files so I can import them into KiCad. Currently I have to save a new file with all the layers and then go through and delete any unwanted information.

dellus, 2021-03-12:
As your feature request might be never be fulfilled, here an easier procedure:

In your multiple layer drawing switch off all layers in the Layer List (the greyed-out eye icon in the top icons bar), then reactivate the wanted layer only  (eye icon leading the wanted layer line, turning green).
Now you see only the contents of the desired layer. Select all and do a Copy (from copy and paste),and specify the reference point, best 0,0. Leave that drawing open.

Now turn to the default "unnamed document", or a previously prepared empty drawing with your special settings (preferably the same as in your original multiple layer source drawing). Paste in the contents you have copied before with the reference point 0,0 again. Then do "Save as" with new name, that of the layer probably.

gwhitney, 2021-05-19:
I wanted to second this request and strengthen it a bit: I recommend that LibreCAD have a Save as... or Export... option that simultaneously saves all the currently visible layers as separate DXF files. It sounds like that would actually help CaseyB as well. Besides KiCad that CaseyB mentions, there are other tools that need layers in separate files for import. I am specifically concerned with Silhouette Studio, the program that drives a Silhouette Cameo cutter. As that's potentially a significant user base for LibreCAD -- I will be teaching a week-long workshop on using LibreCAD for driving cutting machines, for example -- it might well be worthwhile to facilitate transfer from LibreCAD to Silhouette Studio.

Symbian9, 2021-05-21:
> I recommend that LibreCAD have a Save as... or Export... option that simultaneously saves all the currently visible layers as separate DXF files

Just a suggestion, it should be implemented as *"Save Copy as..."* or *"Export..."*.

*"Save as..."* is really not what it should look like.

closed 2023-06-16 by dxli