LibreCAD/LibreCAD #1330: .dxf file won't open in Autodesk Products

DuncanLithgow, 2021-03-16:
This issue is confirmed and copied from Sourceforge https://sourceforge.net/p/librecad/bugs/608/ the attached file has been opened with LC2.2rc2 and resaved.

### Expected behavior
.dxf files from LibreCAD should open in Autodesk Products

### Observed behavior
The attached file will not open and hangs AutoCAD 2018 ( ha! )
The attached file will not open in AutoCAD 2020 / DWG TrueView 2020

The following error was encountered while reading
in PLOTSETTINGS starting at line 11270:
Class separator for class AcDbPlotSettings expected
Invalid or incomplete DXF input -- drawing discarded.

### Steps to reproduce or sample file
See attached file
[Untitled_DXF_problem.zip](https://github.com/LibreCAD/LibreCAD/files/6150438/Untitled_DXF_problem.zip)

### Operating System and LibreCAD version info
Version: 2.2.0-rc2
Compiler: GNU GCC 7.3.0
Compiled on: Dec 31 2020
Qt Version: 5.12.4
Boost Version: 1.65.1
System: Windows 10 (10.0)

DuncanLithgow, 2021-03-16:
The original bug report text

   Files saved from LC won't open in Autodesk products such as DWG TrueView 2020.

   Format 2007/2004/2000/R14 gives this error (different line number):

    The following error was encountered while reading
    in PLOTSETTINGS starting at line 12430:
    Class separator for class AcDbPlotSettings expected
    Invalid or incomplete DXF input -- drawing discarded.

   Format R12 gives this error:

    Unknown group 49 for table on line 1420.
    Invalid or incomplete DXF input -- drawing discarded.

   Same file can be imported to FreeCAD or DXF2GCODE just fine.

lordofbikes, 2021-03-16:
This was already reported here with #1276 and fixed with #1283.
It was merged on Dec. 28. and rc2 was released on Dec. 31., so it should have worked after saving in rc2.
For me it worked right now with latest rc2-7 code and I can open it then with TrueView 2019 successfully.
[Resaved.zip](https://github.com/LibreCAD/LibreCAD/files/6151953/Resaved.zip)

DuncanLithgow, 2021-03-16:
works for me too, wonder what I did wrong last time i checked. sorry for the noise.

closed 2021-03-16 by DuncanLithgow

lordofbikes, 2021-03-17:
No problem, many thanks for testing.