LibreCAD/LibreCAD #1333: Feature Request: Display Name of selected Block in Status Line

WolfgangAndreas, 2021-03-29:
Hi,
i think it would be useful, if the Status Line would display the name of a block, if it gets selected.
I have projects with many blocks with nearly the same name. But i cannot identify, which of them i am currently selecting.
Or do i miss some documentation, where this is described?

Thank you and keep up the good work, which helps me greatly planning my house 👍 

Wolfgang

DuncanLithgow, 2021-04-03:
I've marked this as a feature request and reproducible since I also can't find a way to see the name of the selected block. @lordofbikes can you confirm that this is not possible in the UI?

jbergengh, 2021-11-30:
I tested this one and I'd like to offer a suggestion.

Every time I click on a layer in the layer list, or a block in the block list, my status line gets completely erased, it's replaced by a one line message, replaces the two-line status bar, which shifts my screen down a line, then I have to wait several seconds before it all goes back to normal.  This is really distracting.

The original request was to "display the name of a block, if it gets selected".  I think the intention was to display the block name when selected in the drawing, not from the block list.

My suggestion is either to add a new two-line item to the status bar, similar to the existing "Selected", "Total Length", "Current Layer" and "Grid Status" items, that would say "Selected / Block (name)", or use the existing "Selected" item and replace the running total of selected entities with the selected block name

There would be no need to display the selected layer, as that is already exists in the status bar.