LibreCAD/LibreCAD #1334: LibreCAD crash continously on windows 10

polgia0, 2021-04-02:
### Expected behavior
I do not think my drawing is so complex to causes a problem to the software
### Observed behavior
The program crashes just suddenly closing.  Very unexpected.
### Steps to reproduce or sample file
I zip the file because I couldn't add it in the original format.
[EP.zip](https://github.com/LibreCAD/LibreCAD/files/6249936/EP.zip)
### Operating System and LibreCAD version info
I use windows 10 on a 64-bit machine and I installed the following version of LibreCAD:
LibreCAD-Installer-2.2.0-rc2-8-g8b2b62d1
LibreCAD-Installer-2.1.3

fa201, 2021-04-02:
With the last build on Ubuntu, I have an error message : _Cannot open the file EP.dxf
Please check its existence and permissions._
Same on release 2.1.3.

DuncanLithgow, 2021-04-03:
@fa201 why do you think these issues are related?

fa201, 2021-04-03:
I just checked on Ubuntu if the behavior was the same and reported the results in case it helps.

lordofbikes, 2021-04-03:
This is a bit weird.
On my Ubuntu 20.04 I can open the file and have no crashes, neither with latest rc2-8 nor with fresh built 2.1.3.
But I can force crashes on Windows with both, 2.1.3 and latest rc2-8, by selecting and deselecting some of the bottom arc entities.
Anyhow, there is nor rule for crashing, sometimes soon, after a few clicks and sometimes it needs 40 or more clicks.
I have no debug env on Windows, where the crash happens, only on my Ubuntu, where it doesn't crash.

@fa201 can you post the Qt versions of your 2.1.3 and latest build please.
On my Ubuntu I use 5.12.8.
Windows 2.1.3 was built with 5.4.1 and rc2-8 was built with 5.12.4.

fa201, 2021-04-04:
@lordofbikes  as you can see both are installed on the same PC with the same Qt package. I use the version 2.1.3 to make tests on strange behavior and compare to 2.2.0-rc. For drafting, I always use the 2.2.0-rc.
Note that all packages necessary to build LibreCAD are downloaded from official Ubuntu repo so I do not choose the version.

Version: 2.1.3
Compiler: GNU GCC 7.4.0
Compiled on: May 12 2019
Qt Version: 5.9.5
Boost Version: 1.65.1
System: Ubuntu 18.04.5 LTS

Version: 2.2.0-rc2-8-g8b2b62d1
Compiler: GNU GCC 7.5.0
Compiled on: Apr  2 2021
Qt Version: 5.9.5
Boost Version: 1.65.1
System: Ubuntu 18.04.5 LTS

morf23, 2021-10-01:
I have also a similar problem.

Intel(R) Core(TM) i7-8550U CPU @ 1.80GHz   1.99 GHz
8.00 GB (7.88 GB usable)
Windows 10 Home Single Language
Version	21H1

I was making the file that is attached. Previously I was looking for a fastener library in this [page](https://wiki.librecad.org/index.php/Part_Libraries), and after adding the path to the library folder, I cannot use a line and select another point on the plane, it suddenly crashes as you can see in this [video.](https://youtu.be/wYdUdzY7kUU)

I erased the library path, and I deleted those files. I updated from librecad 2.12 to 2.13, but I have still the same problem.

Version: 2.1.3
Compiler: GNU GCC 4.9.1
Compiled on: Sep 23 2016
Qt Version: 5.4.1
Boost Version: 1.60.0
System: Windows 

I tried other older files, but they seem normal (no crashes) somehow the problem only happens with this specific file.
[equipo-18000rpm.zip](https://github.com/LibreCAD/LibreCAD/files/7270404/equipo-18000rpm.zip)

DuncanLithgow, 2021-10-02:
hi @morf23 could you start your issue as a new bug? It's typically not useful to have several issues in the same report. If the cause is the same then we will link them together. Please add a link here to the new bug report.

I'll be happy to try and open the file on my windows machine from the new bug report.

morf23, 2021-10-03:
@DuncanLithgow I was making another page for my issue as you were suggesting. But I cannot reproduce the crash anymore. Something happened between yesterday and today, that fixed the crashes... totally stable for now, and I didn't do anything noticeable to my system.

The only thing I did was to keep on saving my file as long as it was not crashing.
I don't know if you want me to erase both my comments to keep just the first issue opened.

Noted the methodology for future issues that I might have.

dxli, 2024-05-18:
Closing old issues with 2.2.1 release candidates and 2.2.2 alpha

closed 2024-05-18 by dxli