LibreCAD/LibreCAD #1335: Importing .DWG files error on macOS Catalina

d4mer, 2021-04-05:
Hi All,

I have an issue on macOS 2.2.0 (not Big Sur though). I try to open a DWG and it fails with the following error:

Cannot open the file
/RC Stuff/airplane blueprints/Plano completo Sinbad 3 metros Export.dwg
Please check its existence and permissions.

The file exists for sure as I was able to open it just fine on 2.1.3.

### Expected behavior
Opening .DWG file

### Observed behavior
[image]

### Steps to reproduce or sample file

[image]

[Plano completo Sinbad 3 metros Export.dwg.zip](https://github.com/LibreCAD/LibreCAD/files/6256695/Plano.completo.Sinbad.3.metros.Export.dwg.zip)

### Operating System and LibreCAD version info
Version: 2.2.0-rc2
Compiler: Clang 10.0.0
Compiled on: Dec 31 2020
Qt Version: 5.14.2
Boost Version: 1.71.0
System: macOS 10.15

DuncanLithgow, 2021-04-05:
I can promise it's not just you. AutoCAD has no problems and reports no errors when running an audit. I've converted the file using AutoCAD 2018 to the following formats:
DWG 2018
DWG 2013
DWG 2010
DXF 2018
DXF 2013

And none of those files will open with QCAD, FreeCAD or LibreCAD. So I went right back to DXF r12 and that works in LibreCAD. See attached.

[Plano completo Sinbad 3 metros Export_r12.zip](https://github.com/LibreCAD/LibreCAD/files/6260119/Plano.completo.Sinbad.3.metros.Export_r12.zip)

d4mer, 2021-04-05:
the thing is the file I posted opens fine in 2.1.3

dellus, 2021-04-05:
Does it really open in 2.1.3? It does not for me. Has it been resaved in AutoCAD meanwhile? And may be different now?
When opened in an editor it is not readable mostly. Binary file? Can LibreCAD read binary dwg at all?
I tried to convert with an older Teigha file converter from 2016, got error message. May be the ACAD file version is just too new.

lordofbikes, 2021-04-06:
The file is AutoCAD 2018/2019/2020 and thus not supported by *libdxfrw*, the error message is misleading.
This makes it a duplicate of #1074.

DWG is always binary, but the first few bytes are ASCII and disclose the file format version. In this file it is AC1032.
See https://knowledge.autodesk.com/support/autocad/learn-explore/caas/sfdcarticles/sfdcarticles/drawing-version-codes-for-autocad.html
*libdxfrw* supports AC1006 to AC1027 which means DWG R10 format to AutoCAD 2013/2014/2015/2016/2017.

*libdxfrw* can even read binary DXF, but I can't remember that I've ever seen one.

From the tests by @DuncanLithgow I assume a problem with splines, which are typically used in aeronautic.
The R12 format doesn't support splines, they are replaced by polylines. That may explain why the R12 DXF works.

I tried to read the file with the *libdxfrw* AC1027 reader but this failed too. My assumption is, that even converted versions of this file contain spline types which are not supported yet by *libdxfrw*.

closed 2021-04-06 by lordofbikes