LibreCAD/LibreCAD #1336: Dark theme usability

LongLeafPine, 2021-04-08:
### Expected behavior

When I change my global theme to a dark background the icons should change to light on dark.

### Observed behavior

Icons are black on dark.

### Steps to reproduce or sample file

Global theme: KeepinDarkGreen
Plasma style: KeepinDarkGreen
Application style: Breeze

### Operating System and LibreCAD version info

Operating System: Kubuntu 20.04
KDE Plasma Version: 5.18.5
KDE Frameworks Version: 5.68.0
Qt Version: 5.12.8
Kernel Version: 5.4.0-70-generic
OS Type: 64-bit
Processors: 4 × AMD FX(tm)-4130 Quad-Core Processor
Memory: 7.3 GiB of RAM

Version: 2.1.3
Compiler: GNU GCC 9.3.0
Compiled on: Mar 23 2020
Qt Version: 5.12.8
Boost Version: 1.71.0
System: Ubuntu 20.04.2 LTS

Is there some way that I could help? I am not a programmer, but I am experienced with using and tweaking computers. I would be open to paying some to have this fixed, if that would help.

fa201, 2021-04-08:
@LongLeafPine : please check if your issue is the same as #1144.

LongLeafPine, 2021-04-08:
It looks like it is exactly the same.

I got this email:
Hi there. Could you submit this as a feature request on the issues tracker?

https://github.com/LibreCAD/LibreCAD/issues
IRC: duncan-nz
Twitter: @BIM_DK
so I tried to do just that.

Thanks for your help. Is there anything that I can help with?

fa201, 2021-04-08:
@LongLeafPine 

For the issue I will close as it is a duplicate. Anyway thanks for reporting the issue.

If you are willing to contribute in one way or an other, I suggest you connect on [Zulip chat](https://librecad.zulipchat.com/login/) to get in touch with the contributor.

fa201, 2021-04-08:
Duplicate of #1144

closed 2021-04-08 by fa201