LibreCAD/LibreCAD #1337: R12 Invalid Polyline Import

blablaHoHo, 2021-04-18:
### Expected behavior

It should display a polyline.

[r12-polyline-bug.dxf.zip](https://github.com/LibreCAD/LibreCAD/files/6331727/r12-polyline-bug.dxf.zip)

### Observed behavior

It displays nothing except a point. It is not something wrong with the scale; zooming does not help.

### Steps to reproduce or sample file

This errors occurs even if the DXF file is generated by LibreCAD itself.

Draw a simple polyline. Save the File as R12. Close LibreCAD. Reopen LibreCAD and open the file. You will notice that you cannot see the polyline.

### Operating System and LibreCAD version info

Operating System Fedora 33
Version: LibreCAD 2.2.0-alpha

dellus, 2021-04-18:
Opens for me a polyline of three segments, U-shaped, with no problems on Win 2.1.3 and 2.2.0 rc2-15.
Maybe a problem with your early 2.2.0 alpha Linux version.

dellus, 2021-04-18:
Also on Linux no problem with 2.1.3.
2.2.0 Linux version not available for me.

fa201, 2021-04-18:
On Ubuntu I have an "import error: undefined error" with the last build.
Version: 2.2.0-rc2-15-g76828565
Compiler: GNU GCC 7.5.0
Compiled on: Apr 17 2021
Qt Version: 5.9.5
Boost Version: 1.65.1
System: Ubuntu 18.04.5 LTS

On the same OS with 2.1.3, I have another error: "Cannot open the file. Please check its existence and permissions."