LibreCAD/LibreCAD #1338: Scale according to 2 reference point (no manual calculation needed)

natrius, 2021-04-19:
### Expected behavior

1. Import image
2. Draw reference line (ex. 34 points for 34m)
3. Modify - Scale with reference points
4. Select object(s) to scale (image for now)
5. Select first endpoint of line
6. Select second endpoint of line
7. Approve

Image is now scaled without the need of calculate anything. 

### Observed behavior

1. Import image
2. Draw line in size of known element in image. (ex. the wall is 34m, draw this wall)
3. Measure line in LibreCAD (156)
4. Modify - Scale
5. Select image to scale
6. Select first endpoint of line
7. Windows openes, calculate RV/DV (Reallife-Value / Drawing-Value), in this case 34000[mm]/156=218
8. Approve

There is one additional step needed right now, additionaly i have to calculate. Its not a big problem, but it could be faster and more convenient in my opinion. 

### Operating System and LibreCAD version info

Windows 10, LibreCAD 2.2.0-rc2

lordofbikes, 2021-04-19:
The additional step in observed behavior is the Options window. But this allows to use the tool for various cases.
And in the expected behavior steps I somehow miss how to tell the Scale tool what's the reference and its relation to real word measurement.
So I'd rather see a restriction in your suggestion.

What I do:
- not draw a line, use *Info* tool instead
- not calculate the scale ratio, LibreCAD will do this for me

1. Import image
2. Use *Tools - Info - Distance Point to Point* to measure known Element (34000) in LibreCAD (156)
3. Select image to scale
4. *Modify - Scale*
5. Select reference point
6. Windows opens, enter "34000/156" in *X* field
7. Approve

natrius, 2021-04-19:
Okay, get what you mean. Thought about this and like this:

1. Import image
2. Modify - Scale with reference points
3. Select object(s) to scale (image for now)
4. select first endpoint (reference-point?)
5. select second endpoint
6. Enter how long the distance between these two points is (34000mm for now)
7. Approve

No measurement beforehand is needed at all. It would feel a bit easier in my opinion, it could be implemented additionally as szr (sz = scale, szr = scale reference) so the current one is not changed at all.

dellus, 2021-04-19:
To define a scaling factor via mouse cursor is a handy feature really missing in LibreCAD, and a standard in other CAD packages.
Expected behavior:
1. Select the object(s) to be scaled (may also be an image).
2. Specify a base point (which is to stay where it is).
3. Specify a reference point (which is to be moved). 
4. Specify the target point.
That's it. Similar to the Stretch feature. 

I would like to include an image here for better understanding, but I can't get it done. How do I insert an image here?

fa201, 2021-04-20:
@dellus, you can add image by copy / paste in the text area. It will be converted automatically. Check the preview to be sure.

dellus, 2021-04-21:
@fa201, I have tried it again, with drag and drop, with copy and paste, with different browsers, in WinXP and PuppyLinux, but to no avail. Not even Preview works. Maybe it's something with my provider.

Thanks for your answer.

natrius, 2025-09-18:
Bit late, but @dellus  when you make a screenshot it gets saved in the buffer and you should be able to insert with CTRL-V. Other than that, under the text area when entering a new comment there is a "Pase, drop, or click to add files" that can be used. 

[image]

(made a screenshot and just inserted with CTRL-V). Still think this would be usefull :)

I still think that