LibreCAD/LibreCAD #1340: Delete a point from lines

natrius, 2021-04-23:
### Expected behavior

I have a "Z" shaped line (drawn with 3 "l" lines) and i want to be able to delete the top right point and end up with an "L".

[image]

While i understand this is maybe not that easy with "l", it should definitely be possible to select one point from the line with "pl".

### Observed behavior

I'm not able to select one point at all to delete it, everytime a line is selected. Blue select-box does nothing, green selects both lines. I tried it with "l" and with "pl", its not possible. I am, however, able to move a point around with dragging it, but i fond no intuitive solution for deleting a point.

### Steps to reproduce or sample file

Draw lines with one point in the middle (=2 lines or 2 line-segments), try to select the middle point and try to delete it.

### Operating System and LibreCAD version info

Version: 2.2.0-rc2
OS: Windows 10

(I tried it on top of an imported image and without an image, no change)

dellus, 2021-04-23:
There is no "point" at the corners, where two lines meet, in the sense of an CAD entity, which you can delete.
If you have drawn the Z with "Line" you can drag the upper right corner (connected line ends) to the upper left starting point. This will leave the first line with a length of 0. So it is an entity you can delete then.
If you have drawn the Z with "Polyline", you can use Tool - Polyline - Delete node to delete that "point".

If you seek advice on how to use LibreCAD, the Forum is the appropriate place.
This github - issues is the right place if you have discovered a bug or you have a feature request or a proposal for an improvement .

natrius, 2021-04-23:
I actually thought this was a mising feature so this was a feature request. Never thought about looking into "Tools - Polyline - delete node". (Thanks for showing it is not missing)

In my opinion it should be possible to select a point or a node individually. So i can move it (i know, thats possible already), delete it with `DEL` or something else without having to use the toolbar beforehand.

melwyncarlo, 2021-11-19:
Implementing what @natrius had suggested would be a bit cumbersome. 
You would have to dynamically implement dynamic markers for each of the points in the graphical view.

May I suggest a simpler alternative?
How about you click on a line, and upon pressing the `DEL` key, it deletes that line and connects the broken points together;
and when you press the `SHIFT + DEL` key combination, it deletes that line as usual, without further intuition.

<hr>

So, when you press the `SHIFT + DEL` keys whilst selecting a given line, the following change occurs :
(from the left image to the resultant right image)

[image]

<hr>

And when you press just the `DEL` key whilst selecting the given line, the following change occurs :

[image]

<hr>

If you'd like me to work on this, I'd happily get on with it.

natrius, 2021-11-19:
It sounds nice, but after thinking about it:
How do you determine so you don't end up with ┐ instead of └ ? When deleting the top right point/node its clear it should be an └ and if the bottom-left is deleted its clear it should be an ┐. Or is it on how you started to draw the lines? (or am i simply missing something? :) )

melwyncarlo, 2021-11-19:
Or I'll just decide based on mouse-click coordinates. The line point closest to the mouse coordinates will lead its way, and connect to the other line point. So, basically a solution to your question.

melwyncarlo, 2021-11-21:
While working on your feature today, I realized that lines are discrete entities, 
meaning their not connected to each other, and not related either.

Polyline is the best course of action, because really they're just a collection of lines 
put inside a common container.

In fact, you can convert an ordinary set of lines into a polyline using the 
'Create polyline from existing segments' tool.

`Shift + Del` could not be implemented because `Shift` causes a focus into the command edit box.
So I used `Ctrl + Del` instead. I left the simple `Del` key-press as it as, because all other graphical entities 
use it for deleting themselves.

I started out using the relative zero marker to highlight a given node (determined by the mouse-click).
That worked. But then, there is also the problem of re-using the marker before deleting the node.

So, I set out to create an `RS_Point` entity member within the `RS_Polyline` class (named _highlightedVertex_).
That worked out great. But then a new problem arose.
When de-selecting the polyline by selecting the polyline again, the highlighted vertex disappears.
But then deselecting all via `Ctr-K` or `Esc`, the highlighted vertex persists.

Now, I'm wondering if I should work on _that_, 
or if I should resort to using the marker instead.
Even if the marker is re-used, I will have already saved the highlighted position.
Would that be a acceptable solution?

melwyncarlo, 2021-11-21:
My attempt :

[https://youtu.be/ieTcJlH8VBw](https://youtu.be/ieTcJlH8VBw)

Check out the issue at the end.

melwyncarlo, 2021-11-23:
My second attempt :

[https://youtu.be/4kWNuREhMd0](https://youtu.be/4kWNuREhMd0)

<hr>

I have forgone the idea of using the (graphically) highlighted vertex. So, this tool will use the default relative-zero marker to highlight the selected vertex. I no longer have to deal with de-highlighting the vertex, as was the problem in my first attempt. Keeping it simple.

melwyncarlo, 2021-11-26:
I just realised. Working on your feature request is merely an addendum to what I had posted here. Solved it! Going to bed now. Will post my completed solution tomorrow.

natrius, 2025-09-18:
I never reported back. Selecting is somehow not that easy for me (i guess missing experience) but now with "pl" - "Delete node" i can simply delete on point of the drawn Z and it changes the way i expected it. 

[image]

closed 2025-09-18 by natrius