LibreCAD/LibreCAD #1342: Build error with boost 1.76

rezso, 2021-04-30:
Building librecad 2.1.3 release with boost 1.76 fails:

`lib/engine/rs_ellipse.cpp:70:15: error: ‘tuple’ in namespace ‘boost::math’ does not name a template type
   70 |  boost::math::tuple<double, double, double> operator()(double const& z) const {
      |               ^~~~~
In file included from /usr/include/boost/config.hpp:61,
                 from /usr/include/boost/numeric/ublas/matrix.hpp:17,
                 from lib/math/lc_quadratic.h:30,
                 from lib/engine/rs_ellipse.cpp:38:
/usr/include/boost/math/tools/roots.hpp: In instantiation of ‘T boost::math::tools::halley_iterate(F, T, T, T, int) [with F = {anonymous}::EllipseDistanceFunctor; T = double]’:
lib/engine/rs_ellipse.cpp:136:23:   required from here
/usr/include/boost/math/tools/roots.hpp:681:65: error: no match for call to ‘({anonymous}::EllipseDistanceFunctor) (double)’
  681 | inline T halley_iterate(F f, T guess, T min, T max, int digits) BOOST_NOEXCEPT_IF(policies::is_noexcept_error_policy<policies::policy<> >::value&& BOOST_MATH_IS_FLOAT(T) && noexcept(std::declval<F>()(std::declval<T>())))`

Tarot67, 2021-05-01:
OI have the same error I found solution at this problem
https://www.boost.org/doc/libs/1_73_0/libs/math/doc/html/math_toolkit/internals/tuples.html
Just add in librecad/src/lib/engine/rs_ellipse.cpp
add this include:
#include <boost/math/tools/tuple.hpp> 
after 
#include <boost/math/tools/tuple.hpp>

and it work's

lordofbikes, 2021-05-19:
Fixed with #1345

closed 2021-05-19 by lordofbikes